prusa3d/PrusaSlicer #3240: Infill Percentage Not Accurate with Extrusion Widths Expressed as Percentages

klowell, 2019-11-26:
### Version
PrusaSlicer v2.1.0

### Operating system type + version
Windows 10 (64-bit)

### 3D printer brand / version + firmware version
Creality Ender 3X

### Behavior
When extrusion widths are expressed as percentages instead of millimeter values, the infill is much too tight.

Under Advanced Tab of Print Settings, express extrusion widths as millimeter values.
Slice
Note that infill percentage is correctly proportioned.
Now, change extrusion widths as percentage
Re-slice
Note that infill is much tighter

Found this out when using 3D Honeycomb infill pattern.

lukasmatena, 2019-11-26:
Could you please enclose a 3MF? I cannot reproduce the problem.

supermerill, 2019-11-26:
> extrusion widths are expressed as percentages

It's a % of the layer height. You are sure it's what you want?

klowell, 2019-11-26:
Sorry, it was a long night last night, and I thought it was assumed over nozzle diameter even though the dialog clearly says layer height.  I'm going to go ahead an close this one now, thanks.

closed 2019-11-26 by klowell

supermerill, 2019-11-27:
I agree it's not very intuitive. It seems that it's a relic from the past when extruders have a constant flow, or something like that.
I changed that in my fork to be % of (the first) nozzle diameter but I understand that prusa don't want to break any user configuration by modifying the slicer behaviors.