prusa3d/PrusaSlicer #3241: Modifier using infill overlap has no effect.

yru2501, 2019-11-26:
### Version
2.1

### Operating system type + version
Ubuntu Appimage
### 3D printer brand / version + firmware version (if known)
golemD - but issue is obvious even in slic3r preview :)

### Behavior
if you set general overlap it will remain regardles of block/part/model/layer height modifiers

So you can't for example increase overlap for some region or height where you need extra strenght
And it goes back to at least 1.29 slic3r :)

supermerill, 2019-11-26:
I think the bug is due to how prusaslicer merge surfaces. It's not consistent, and some parameters are completely ignored.

I didn't took the time myself to check every settings, I think i/somone have to create a unit test per setting to see if all of them are taken into account.

rtyr, 2021-04-27:
Should be fixed. Closing.

closed 2021-04-27 by rtyr