prusa3d/PrusaSlicer #3242: No "custom gcode" field in height range Modifier

yru2501, 2019-11-26:
### Version
2.1

Custom g-code field  triggered at given height from range modifier could open myriad of possibilities if properly saved in 3mf file.

lukasmatena, 2019-11-26:
Is this super-detailed feature request a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3057? If so, there is a myriad of other similar duplicates referenced from there.

yru2501, 2019-11-26:
If it will be possible to define unique g-code for every pause then yes. Range modifier approach seemed more intuitive. It would also probably trigger the injection at start of this particular parts section of g-code. But I'm just dropping an idea.

rtyr, 2020-04-22:
Implemented, see https://github.com/prusa3d/PrusaSlicer/issues/3057. Closing.

closed 2020-04-22 by rtyr