prusa3d/PrusaSlicer #3244: Octoprint Settings when Opening a .3MF file

sgano, 2019-11-26:
### Version
V2.1.0

### Operating system type + version
MacOS 10.15.1 

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
When opening a .3MF file the Octoprint settings API / host settings are not preserved.  This is probably a good thing for security when sharing files.   However, this also means you can't make setting changes and then quickly re-upload the file to Octoprint when used locally.   This results in only being able to send a file to Octoprint directly when the file is initially created and not after a file has been saved and re-opened. 

There should be a way to globally set your Octoprint settings for a particular profile (and if a file is loaded that inherits from that profile, then load the local Octoprint settings).   Or some other way to specify the Octoprint settings so they can be used on re-opened files.

This is slightly different than issue #3050 - which is similar in the request for global Octoprint settings - but this is to prevent having to re-enter this data when the system present profiles are updated by Prusa.  This is also a specific issue of the more generic #3238 . 

#### Project File (.3MF) where problem occurs
NA

Jebtrix, 2019-11-26:
Current workaround see issue https://github.com/prusa3d/PrusaSlicer/issues/3050

Once you do that and open a saved .3MF file, you will see the Orange circle arrow icon, if you click it will restore defaults (which now is your manually set octoprint defaults for all printers). Not click free, but a single click beats copy pasting.

sgano, 2019-11-27:
Thanks @Jebtrix ! I was missing the part of clicking on the lock to restore defaults.    Though on my Mac this still doesn't work; but it probably is  because I am editing the wrong .ini file though.   I have tried editing:

PrusaSlicer.app (right click, select "show contents")  -> Contents/Resources/profiles/PrusaResearch.ini

Jebtrix, 2019-11-27:
Should be located at ~/Library/Application Support/PrusaSlicer/vendor

sgano, 2019-12-02:
That was the correct PrusaResearch.ini;  thanks again @Jebtrix !

bubnikv, 2020-02-18:
patched with 25d6818fc7351ff008674adb8ff15e9522246764

When loading a configuration file (from AMF, 3MF, .ini or .gcode), 
and if the host settings ("print_host", "printhost_apikey", "printhost_cafile")
in the configuration file is empty, then the config is considered to be anonymized,
and these anonymized keys are loaded from the referenced "derived from" profile,
if it exists in user's PrusaSlicer configuration directory.

We are aware that this is a patch work and a better long term solution
is to separate physical printer settings from logical printer settings,
but this is a good enough solution for PrusaSlicer 2.2.0 release.

The patch will be part of PrusaSlicer 2.2.0-beta. Closing.

closed 2020-02-18 by bubnikv