prusa3d/PrusaSlicer #3247: Feature Request:  Blip Fan When Turning On From Idle

klowell, 2019-11-26:
### Version
PrusaSlicer 2.1.0+win64-201909160915

### Operating system type + version
Microsoft Windows 10 (64-bit)

### 3D printer brand / version + firmware version (if known)
Creality Ender 3X

### Behavior
This is a feature request to add a "blip" to the part(s) cooling fan option.  This feature is on Simplify3D and helped me with one of my prints.  To my understanding, a "blip" sets the fan to 100% for a very short amount of time in order to get the fan spinning.  For some fans, the motor inside the fan does not have enough torque at low fan speed to overcome the static friction when the fan was left idle, but will have enough torque to keep the fan moving.  

So, if you want to start the fan at say 25%, the fan will not spin, as the 25% power being sent to the fan is not enough to get it spinning.  A "blip" will help overcome this by overcoming that initial friction with a burst of full power.

I print mostly in ABS on my Ender 3 with its stock cooling fan, and discovered this problem when I was printing Hatchbox ABS; which does need cooling uses 25% fan speed.  I found that my fan never turned on its own, I had to start the fan manually with my finger, and after that I never had to start with my finger again.

Apologies if this is a feature already (and I could not find it) or if someone else has requested it in the past.

Jebtrix, 2019-11-26:
Unfortunately it looks like devs want to leave it to firmware 
* Feature Request : Blip fans to full speed: https://github.com/prusa3d/PrusaSlicer/issues/327

italocjs, 2020-02-04:
Has this ever been added? its not such a good idea to leave it to firmware (since only marlin supports, and the user has to have access to the source code and know how to code it to add the option to the firmware, which most printers manufacturers dont release), for now i'm using a post-script to do it, but its not the best approach (simplify 3d has a "blip to full" box that does this)

pedrobecerra, 2023-11-23:
Just have a look to Simplify3D. It has this option.
[image]

pedrobecerra, 2023-11-23:
Just have a look to Simplify3D. It has this option.
; layer 3, Z = 0.600
M106 S255
G4 P500
M106 S51
[image]

closed 2026-09-29 by prusaslicer-bot[bot]