prusa3d/PrusaSlicer #3248: Artifacts generated at pure tower when 'Single Extruder Multi Material' deselected

joris57, 2019-11-26:
Version 2.1.0+win64
Windows10
Printer : N/A 

Behavior
When slicing a multi-material object for a printer with 'Single Extruder Multi Material' deselected, unexpected artifacts are generated at the purge tower.

[image]

[Moai.zip](https://github.com/prusa3d/PrusaSlicer/files/3894258/Moai.zip)

Jebtrix, 2019-11-26:
Intentional, see issue Purge block printing in air https://github.com/prusa3d/PrusaSlicer/issues/3116

joris57, 2019-11-26:
The referenced issues talk about an extra ramming at the end of the print. In the case described in this issue, it happens at the start of the print (3 times) and in he middle (1 time) and NOT at the end of the print.

Furthermore, single extruder multi material is deselected, so I would not expect any ramming to occur.

Jebtrix, 2019-11-26:
Yeah list of other issue references do seem to happen at the end. This file slices the same in latest git master also.

lukasmatena, 2019-11-27:
Thanks. This might be somehow related to #3234 and maybe #2968. I'll investigate.

joris57, 2019-11-27:
Additional observation to maybe help determining the source of this issue : 

With 'Single extruder Multi Material' selected, the same artifacts appear when setting the 'Total ramming time' to 0.

lukasmatena, 2019-12-05:
There really was a bug regarding the move to the wipe tower. This should now be fixed by 
4ba7cb7 as @jmd73 confirmed in #2968. This issue is describing the same and I tested that current master does not produce the extra extrusions with your 3MF. The fix will be in upcoming PrusaSlicer 2.2 release.

Closing the issue, thanks for the report.

closed 2019-12-05 by lukasmatena