prusa3d/PrusaSlicer #3249: First support layer is always rectilinear regardless of setting

drewbus-drewbus, 2019-11-27:
### Version
2.1.0 Win64

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S
### Behavior
No matter what support setting I select, the first layer is always rectilinear. This causes large print to fail due to bed adhesion issues.

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-03-15:
because it is using the bridging flow.

bubnikv, 2021-02-24:
I don't understand what is the problem with the rectilinear pattern for the 1st support layer.
Anyways, the current master allows one to set the 1st support layer density and it draws a single perimeter around the 1st support layer, which should improve bed adhesion.

closed 2021-02-24 by bubnikv