prusa3d/PrusaSlicer #3251: Reset project name when plate is empty

apphead, 2019-11-27:
### Version
2.1.0 win64

### Operating system type + version
Win 8.1 x64

### FEATURE REQUEST
Since most users will not create a new instance of PS for every print, but instead load new objects into PS, a new file name should be used. 
ALWAYS Remove project / document **name** if plater is empty. 

For clarification: project or document name is the one that is displayed in the window title and the filename that is suggested when exporting gcode.

* _Describe the problem_

* _Steps needed to reproduce the problem_
- Open any .3mf file in PS
- Remove each object individually by hitting _**Del Key**_  or _Ctrl + A_ then _Del_
- Drag any STL into the same instance of PS
- Save project by _Ctrl + S_

==> project name is retained

* _Expected Results_
Ctrl + S will not overwrite _previous_ project but will ask for a new filename 
Project name is reset once the plater is completely empty - irrespective of how you empty the plater.
If plater is empty the application status should be the same as if you cleared the plater by clicking on the trash bin icon in the top bar (or _Ctrl + Del_ respectively) 

* _Actual Results_
Previous project name is reused. Which causes a new project to wrongfully override the previous .3mf

_Is this a new feature request?_
YES

bubnikv, 2020-03-16:
We consider the correct behavior correct. If you click "Delete all", the project name is cleared, while if you delete the objects one by one, the project name is maintained. That is by design.

closed 2020-03-16 by bubnikv

apphead, 2020-03-16:
Ok, you're the boss. But still and as I layed out 
> Project name must be reset once the plater is completely empty.
I believe it's a fairly safe assumption that once the plater is empty, a new project is started. Or then one would / should have chosen to reload object and continue with the existing projectname.
It's a minor thing but had lead me to wrongfully overwrite existing .3mf files by just ctrl-s saving.

bubnikv, 2020-03-17:
I will ask our co-workers for their opinion.

foreachthing, 2020-03-17:
How about making it optional? No one looses.