prusa3d/PrusaSlicer #3252: Include screenshot / thumbnail of plater in .3mf file

apphead, 2019-11-27:
### Version
_2.1.0 x64_

_feature request_
As outlined in [](https://3mf.io/3d-manufacturing-format/) it would be nice to add a thumbnail to the zip. Maybe some OS decide to extract and display it the file manager!?

In terms of useability, if Prusa decided to create a thumbnail viewer for .3mf in Win / Linux / Mac file exploerer, well, that would be truly awesome.

enricoturri1966, 2019-11-28:
This feature is already in current master.

apphead, 2019-12-11:
Tested new feature with 2.1.1 but there is no thumbnail contained in the .3mf(as specified here: https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#61-thumbnail )
Tested by re-saving an existing project and from scratch. The files did not differ from previously generated .3mfs

enricoturri1966, 2019-12-11:
This feature is not included into 2.1.1
It will be released with 2.2.0

apphead, 2019-12-11:
@enricoturri1966 isn't that what is said in the release notes for 2.1.1:

> and 3D scene thumbnails are newly generated and exported into the G-code to be displayed on the Prusa Mini printer's display.

PS: I don't want to rush you, just thought I'd mention it since it was announced and I opened a ticket
PPS: how about closing a few of those 1k+ tickets anyway (a test manager's nightmare 8-) )

enricoturri1966, 2019-12-11:
>> isn't that what is said in the release notes for 2.1.1:
No, it is not. It says that the thumbnails are saved into the gcode, to allow the Prusa Mini printers to show them on their display. That's it. There is no mention to 3mf files ;)
On current master, thumbnails are exported also to 3mf and to sl1 files. These exports will be part of upcoming release of version 2.0.0 of PrusaSlicer.

apphead, 2019-12-11:
> There is no mention to 3mf files ;)
True. So Prusa Mini includes a G-code viewer then. Nice!
So I will check back when 2.2.0 is released.

bubnikv, 2019-12-20:
2.2.0-alpha1 will be released any time soon. Closing.

closed 2019-12-20 by bubnikv