prusa3d/PrusaSlicer #3253: Cooling defined by mm^3 instead of seconds

espr14, 2019-11-28:
v2.1.0 win64

Cooling thresholds are defined by seconds per layer. This is intuitive but not physically based. If I print 0.1 or 1 mm layers the cooling would be the same but more coooling is needed for 1 mm layers. The same applies to the nozzle diameter.

Cooling thresholds should be in mm^3 of filament used per layer.

joris57, 2019-11-29:
How is the 'cooling time' (a.k.a) 'layer print time' taken into account in your suggested alternative? A 10mm3 layer printed in 5 sec may require more cooling or slowing down where a 10mm3 layer printed in 50 sec would perfectly ok at the set speed and/or fan levels.

espr14, 2019-11-29:
You're right. It should be in mm^3/s. More volume and shorter time requires more intensive cooling.

joris57, 2019-11-30:
mm3/s is the (average) volumetric flow. Given two prints with the same volumetric flow, a large surface will have more time to cool than a small surface, requiring increased cooling/layer print time for the small, but not for the large surface, so I don't think mm3/s provides an better alternative.

espr14, 2019-12-02:
There are two types of cooling: active (fan) x passive (ambient air).

- Active: it depends on time spent at the place which is defined by speed (mm/s and size of fan duct size) and on crosssection size of thread (mm^2). So the result is **mm^3/s**.
- Passive: it depends on time per layer and on thickness of layer (mm). So the result is **mm^2/s**.

joris57, 2019-12-02:
Sorry, but you completely lost me there. Can you summarize from the above how you think fan speed and/or print speed should be controlled with respect to layer cooling?

espr14, 2019-12-02:
Keep fan off or running slow until passive threshold is met (in mm/s). Then activate fan linearly up to the limit defined in mm^3/s. Then slow down print up to the minimal print speed (mm/s).

joris57, 2019-12-02:
Sorry, I give up. I'm to stupid to understand ...

espr14, 2019-12-03:
[image]

joris57, 2019-12-03:
There is no such thing as 'maximal coolable volume speed'. ANY (realistic) volumetric flow can be cooled (by ambient air) IF the layer (and hence the layer time) is large enough.

It would also be nice to see the logic expressed in pseudo-code or in plain text as currently available in Prusaslicer.

espr14, 2019-12-05:
Total cooling (TC) = passive cooling (PC) + active cooling (AC). Let's say we want to achieve TC >= 1.

Logic:
1. If PC>1 then turn off fan for whole layer (except bridges etc.).
2. If PC<1 then turn on fan with enough speed so PC+AC>1.
3. If PC+AC<1 then decrease speed so PC+AC>1.

Formulas:
PC = (time_of_layer / layer_height) * magic_constant_PC
AC = 1/(thread_cross-section_area * print_speed) * fan_speed * magic_constant_AC

joris57, 2019-12-06:
The parameters that you defined in your mock-up user interface don't seem to appear anywhere in your algorithm, so where are these used?

How are the 'magic constants' defined or determined?

espr14, 2019-12-06:
Here's the new GUI:
[image]

PC constant is easy:
magic_constant_PC = typical_layer_height / fan_below_layer_time

AC constant has to be defined by tests. If the fan is too slow then the magic_constant_AC has to be decreased.

joris57, 2019-12-06:
I'd like to have a go at quantifying the algorithm. Starting with magic_constant_PC ...

What would these parameters be for 'generic PLA'? I guess typical_layer_height would be 0.2mm, but what is 'fan_below_layer_time'? Fan is always at 100% for this filament profile, regardless layer time.

joris57, 2019-12-07:
With respect to your logic:

1. If PC>1 then turn off fan for whole layer (except bridges etc.).
2. If PC<1 then turn on fan with enough speed so PC+AC>1.
3. If PC+AC<1 then decrease speed so PC+AC>1.

I assume that (2) and (3) should read PC+AC=1, because if it would be PC+AC>1, then the question is 'how much larger' ????

espr14, 2019-12-09:
We are interested in auto cooling only. You can keep fan always on but that just decreases strength of large objects which are overcooled. If you properly set auto cooling you don't need fan always on turned on.

For typical usage magic_constant_PC = layer_height/time_of_layer = 0.2/100 = 0.002. The numbers are very low so I would use reciprocated value = 100/0.2 = 500.

magic_constant_AC =  (thread_cross-section_area * print_speed) / fan_speed = (0.45*0.2 * 45) / 1 = 4.05. This number is low so it would be presented in % => 405 %.

Let's make the algorithm more precise:
```
if (PC>1) fan_speed(0);
else {
  fs = ((1 - PC) * (thread_cross-section_area * print_speed)) / magic_constant_AC;
  if (fs<1) fan_speed(fs);
  else lower_print_speed();
}
```
Current Marlin uses byte for fan speed so the number has to be multiplied by 255. Computing print speed multiplier is tricky because slower print influences both AC and PC. On the other hand, non-print movements affects only PC. So there should be some iterative algorithm which in few iterations computes some values so the PC+AC>1.

DrKropp, 2019-12-12:
I like this idea. I change filament types and nozzle sizes quite often. By linking the cooling fan speed to the volumetric flow I could spend much less time tuning the cooling settings. 

Another idea would be to link fan settings to the degree of overhang. Vertical walls need less cooling than overhangs, especially near tight corners. It might help prevent curling in some models.

espr14, 2019-12-12:
Overhangs need 100% fan speed typically so you have to tune extrusion rate and print speed more likely.

rongith, 2019-12-18:
@espre14, no overhang don't need 100% fan in any cases. Cornered overhangs don't like fan cooling as there is then too much curling up due to internal stress.

espr14, 2019-12-18:
It depends on the approach to overhangs. Also it depends on printer since they have different fan flow volume and different type of fan nozzles.

danoveb, 2022-04-24:
I really like this idea. I'm using my printer to only make functional parts in PETG, and have also realized that the automatic cooling should be based on filament flow but what I think is missing here is the concentration of extrusion in the same area. Take the chimney on 3Dbenchy as a good example. It will deform if I don't have cooling on, but at the same time, I don't want cooling on single perimeter hole/pipe with the same diameter that have shorter layer time... I'm far from an expert, but there should be some setting filament volume/space volume controlling the amount of cooling needed.

danoveb, 2022-04-24:
A simplified alternative is just to be able to set the cooling proportional to the layer print area in the same way we set the thresholds for layer time today. The problem with current solution is that we can have a one perimeter pipe/hole that takes a Couples of seconds to print and then triggers the fan threshold, but if we have the same pipe with 3 perimeters and maybe infill, it will take a lot more time and not trigger the fan threshold and it will deform.

danoveb, 2022-05-08:
I have now  made a simple post script that just takes the min and max of both X and Y in every layer, calculates the area and sets a interpolated fan speed from 4 parameters: min and max area parameter / min and max fan speed. This is a very simplified solution, but works quite well.

Tupson444, 2026-09-12:
still relevant