prusa3d/PrusaSlicer #3254: Simple model produces empty layers

gibell99, 2019-11-28:
### Version
2.1.0 Win64

### Operating system type + version
 - OS: Windows 10 pro
 - Version 1803
 - OS Build 1734.1006

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s

### Behavior
1. Import stl 'puzzlecad-gontier2-1.stl'
2. Slice
3. Notice empty layers in part of the model (see screen shot)
4. This may be a problem with the stl model, but it seems OK.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
If I save the file as a 3MF file and reopen and slice, it seems to work.  Also "Split to Objects" and then slicing also works.

[image]
[image]
[puzzlecad-gontier2-1.zip](https://github.com/prusa3d/PrusaSlicer/files/3902743/puzzlecad-gontier2-1.zip)

Jebtrix, 2019-11-29:
The model has a couple mesh issues including non-manifold. Since your on Win10 you should just right click on model and select Fix Through Netfabb to resolve it.

gibell99, 2019-11-29:
Thank you for your comment.  I agree that the model seems to be in error.  It would be nice if Slicer at least gave some sort of error message to warn us.  I notice it sometimes reports fixing an stl file, but it doesn't say anything in this case.  Even reports: "Manifold: Yes".

Jebtrix, 2019-11-29:
Well manifold is one type of mesh issue. There is zero length edges, zero faces, intersecting faces, non-planar faces, and the list goes on. The fun part is usually you can't tell by just looking at the mesh. Merging vertices at a distance ends up cleaning up some of those issues (not manifold related).

smurfix, 2019-12-04:
It's not a model problem. I've seen this error with a perfectly valid small dome, from https://cdn.thingiverse.com/zipfiles/03/96/73/65/f6/Calibration_dome.zip
I scaled the STL to 1000% (the original is a millimeter high, very funny) and got the error for z: 1.400.

I imported the thing in Blender and there definitely are no gaps, de-normalized faces, or in fact any other error I can see in this thing.

lukasmatena, 2019-12-05:
@smurfix Could you enclose a 3MF? I opened that stl, scaled to 1000 % and didn't see anything wrong when sliced using the system MK3S profile.

Jebtrix, 2019-12-06:
> 
> 
> @smurfix Could you enclose a 3MF? I opened that stl, scaled to 1000 % and didn't see anything wrong when sliced using the system MK3S profile.

I didn't see anything wrong either at %1000. The model itself just has duplicate vertices and 1 zero/null face.

smurfix, 2019-12-16:
Reproducing this problem is easy, just set the number of perimeter layers to zero.
Slicer should either support that option, or have a minimum of 1.

lukasmatena, 2019-12-17:
@smurfix Could you enclose a 3MF? I opened that stl, scaled to 1000 %, set number of perimeters to zero and didn't see anything wrong when sliced using the system MK3S profile.

I don't know what the problem is but I apparently need a 3MF to reproduce it. Was it really so unclear in my last post?

smurfix, 2019-12-17:
Sorry, I was too hectic as usual.

Howeever, fortunately(?) I can't reproduce the problem with the current master, so I'd recommend to close this issue.

bubnikv, 2021-09-23:
> However, fortunately(?) I can't reproduce the problem with the current master, so I'd recommend to close this issue.

Cool, closing.

closed 2021-09-23 by bubnikv