prusa3d/PrusaSlicer #3255: Parameter Name "extruders_count" is not available for use in Custom G-code

ld3300, 2019-11-29:
### Version: 
PrusaSlicer 2.1.0
### Operating system: 
Windows 10 Home Version 1903 OS Build 18362.476

### 3D printer: 
Creality Ender 3 Pro modified with MKS Gen L with 5 x TMC2208 Stepper Drivers
### Firmware: 
Klipper Git version: 'v0.7.0-803-g1b454c0-dirty'

### Behavior
Attempting to use the Parameter Name "extruders_count" from Printer Settings > General > Capabilities > Extruders: returns an error after slicing that the variable does not exist. (Caret symbol lines up with the 'e' in "extruders_count" when viewed in Notepad++).  The code is trying to pass the value of extruders_count to a macro in Klipper called GET_EXTRUDERS. Full configuration zip included below.

### Expected Result: 
extruders_count should be available to Custom G-Code commands in PrusaSlicer.

### Actual Result: 
From gcode.tmp output file (full file included below):
> !!!!! Failed to process the custom G-code template start_gcode
> Parsing error at line 3: Variable does not exist
> GET_EXTRUDERS S[extruders_count]
>                 ^
> !!!!! End of an error report for the custom G-code template start_gcode

[image]

[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/3903905/PrusaSlicer_config_bundle.zip)

[15488.gcode.tmp.zip](https://github.com/prusa3d/PrusaSlicer/files/3903922/15488.gcode.tmp.zip)

supermerill, 2019-12-17:
It's not a "real" setting. 
The number of extruders is not stored, what is stored is all the idx of the used extruders. This field is only used in the gui to resize gui things (and settings arrays).
The number of extruder is something that is gathered at gcode generation, because some may be useless. 
It's possible and easy to add the setting at gcode generation, search for "m_placeholder_parser.set" and create a new line with "m_placeholder_parser.set("extruders_count_used", tool_ordering.all_extruders().size())"
I don't know what your macro is trying to do, you have a variable number of extruder?

ld3300, 2019-12-22:
Thank you, I will try to wrap my head around that.  I have a klipper macro that will decide how to treat my dual extruder, single hot end setup based on number of extruders from the slicer (like don't prime the second extruder, or set it up to automatically take over on filament runout sensor event)

closed 2019-12-22 by ld3300