prusa3d/PrusaSlicer #3257: Slow startup Ubuntu 19.10

WZ9V, 2019-11-30:
Version: 2.1.0+linux64
Build: PrusaSlicer-2.1.0+linux64-201909160925
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 5.3.0-23-generic x86_64

There is a very long pause when starting PrusaSlicer on Ubuntu 19.10.  I didn't see this before on Linux Mint which I believe is based on Ubuntu 18

WZ9V, 2019-12-02:
Went back to Ubuntu Desktop 18.04 and the startup was normal again.

AmyTheCute, 2019-12-05:
I want to same the exact same happens on 19.1. haven't tried LTS but it works fine on windows.
app functions and works normally and smoothly after it opened

killerb77, 2019-12-09:
I can confirm the same.  Ubuntu 19.10 is very slow to open.  18.04 was fine.  I also tried Pop OS 19.10 and it opened slow as well.  This is the appimage.  I was going to try to compile from source code but I kept running into dependency issues and couldn't solve a few of them so I couldn't see if compiled on my system would be any different than the appimage.

AmyTheCute, 2019-12-11:
I really wish we had a .deb of a flatpak or something. the appimage doesn't even have an icon

kelchm, 2019-12-30:
Also seeing the same issue on Pop OS 19.10 with either the Appimage or the extracted archive.

stephbur, 2020-04-04:
This is happening for Ubuntu 20.04 preview as well with the Appimage.

stephbur, 2020-04-04:
After having the same issue with Okular (startup times ~30 secs), I found this stackexchange post:

https://askubuntu.com/questions/1184774/some-applications-on-ubuntu-19-10-very-slow-to-start

The gist:

`sudo apt-get install appmenu-gtk2-module `

After a reboot everything was working blazingly fast again. 
I would suggest to add the above package to the dependencies for gtk based systems.

xarbit, 2020-04-27:
> I really wish we had a .deb of a flatpak or something. the appimage doesn't even have an icon

Hi @AmyTheCute,
there will be a flatpak build on flathub in the next hours. 
https://github.com/prusa3d/PrusaSlicer/issues/1124

rtyr, 2021-02-04:
Is this issue still valid? Does it happen with 2.3.0?

bubnikv, 2021-02-06:
> Is this issue still valid? Does it happen with 2.3.0?

I believe this issue is a platform bug, not PrusaSlicer bug.
On those systems you most likely may run PrusaSlicer-2.3.0 built against GTK3 and the issue will go away.

xarbit, 2021-12-12:
@bubnikv @lukasmatena this can be closed, or?

closed 2021-12-13 by bubnikv