prusa3d/PrusaSlicer #3258: Simple model can't be sliced, claims there are no layers. Cura slices it fine.

mjparme, 2019-11-30:
### Version
PrusaSlicer
Version:   2.1.0+
Build:     PrusaSlicer-2.1.0+-201909160907

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.15.1 (Build 19B88)
Total RAM size [MB]: 25,770MB
OpenGL installation
GL version:   2.1 ATI-3.2.24
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 570 OpenGL Engine
GLSL version: 1.20

### Operating system type + version

Mac OS 10.15.1 (Catalina)

### 3D printer brand / version + firmware version (if known)

### Behavior

I have a simple model I created in OpenSCAD that PrusaSlicer says has no layers and it refuses to slice the model. This same model slices fine in Cura. Since I created this in OpenSCAD it should be manifold (from my understanding you can't create non-manifold models in OpenSCAD).

[image]

[honeycomb-2inch-cut-template.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3907595/honeycomb-2inch-cut-template.3mf.zip)

[honeycomb-2inch-cut-template.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3907598/honeycomb-2inch-cut-template.stl.zip)

mjparme, 2019-11-30:
After creating this issue I discovered that if I save the project as a 3mf, then import the 3mf it can slice the model just fine. However, if I import the STL directly then try to slice it I get the error. Weird.

wabbitguy, 2019-11-30:
I loaded your STL into 2.1.0 and found the same thing. Opened it in Netfabb, which said there were no errors, repaired it there anyway, saved the repair, loaded the new STL into Slicer and it slices perfectly.

Odd...very odd...

mjparme, 2019-11-30:
MeshMixer also doesn't show any errors in it. Also, very odd you can just "sava project as" to a 3mf. Then immediately import the just created 3mf and it will slice it.

Jebtrix, 2019-12-01:
The mesh has some zero faces and zero edges. Merging vertices at distance of 0.0001 world units eliminates the zero features and it will slice correctly. I feel like a broken record saying this :D 

If saving as 3MF fixes it, then the export code is smart enough not to propagate the zero features. Which is good then devs already have the solution, per se.

mjparme, 2019-12-01:
Here is some OpenSCAD code that puts two cubes together without merging them slightly into each other. PrusaSlicer reports it as manifold and can slice it. 

```
size = 5;
cube([size, size, size]);
translate([size, 0, 0]) cube([size, size, size]);
```

resulting stl: 
[sandbox.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3908755/sandbox.stl.zip)

smurfix, 2019-12-05:
@mjparme What do you mean, "without merging them"? the STL you uploaded is a single 10x5x5 cube.

mjparme, 2019-12-05:
@smurfix an earlier comment said with OpenSCAD you have to slightly merge two objects to fix the problem of zero features "Merging vertices at distance of 0.0001 world units eliminates the zero features and it will slice correctly." My example is showing that you don't have to do that to get a manifold object. The end result is a 10x5x5 cuboid; however, it was created with two 5x5x5 cubes. I did not merge them into each other by a very small amount and the object is Manifold and PrusaSlicer slices it without issue.

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo