prusa3d/PrusaSlicer #3259: Slicing hangs on generating perimeters with thing:3565827 (30g)

computer1up, 2019-12-01:
### Version
2.1.0
### Operating system type + version
macOS 10.14.6 Mojave

### 3D printer brand / version + firmware version (if known)
MK3S - 0.30mm draft profile
Generic PLA profile
### Behavior
Slicing the 30g file from thing:3565827 (thingiverse.com) (DIY Multi-color filament) results in PrusaSlicer hanging on generating perimeters. CPU usage for PS is almost 0%. 
This is a file with a very long perimeter. Therefore it will never actually slice the file. Cancelling the slicing job results in an overall hang of the program. 

#### Project File (.3MF) where problem occurs
[Fillament_30g.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3908576/Fillament_30g.3mf.zip)

bubnikv, 2020-03-15:
usually the "avoid crossing perimeters" feature is the culprit. Is it in your case?

computer1up, 2020-03-15:
In the attached 3mf file, this seems to be turned off. I'll try with and without.

FidelCapo, 2020-03-16:
Thank you for report. I was able to slice this 3mf but it took me 18 min. You are right about canceling that results in hang of the program.

bubnikv, 2020-03-16:
PrusaSlicer hates spirals?

[image]

bubnikv, 2020-03-16:
This is certainly something to look into for PrusaSlicer 2.3.

bubnikv, 2020-03-16:
Gap fill is the culprit. Disabling the gap fill by setting its speed to zero solves this particular issue.

[image]

Gap fill is calculated from inside the perimeter generator. Voronoi diagram is built for what remains after the perimeters are extruded, and for this particular model the Voronoi diagram is so complex, that the boost::voronoi algorithm is at it knees.

computer1up, 2020-03-16:
Thanks guys! I haven't been able to test it further yet, but I'll try it with the gap-fill setting turned off for now. Good thing we at least now know what the problem is.

bubnikv, 2022-11-16:
It will be fixed in PrusaSlicer 2.6.0-alpha1.
Closing.

closed 2022-11-16 by bubnikv