prusa3d/PrusaSlicer #3262: Enhancement suggestion: Since different filament materials shrink at different rates, consider adding a scaling option to the "filament settings" tab.  This would enable more accurate printing.  Thanks, John.

OldSalt1945, 2019-12-02:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

MStylesMS, 2020-10-30:
I agree with this request.  Having a default shrinkage factor for each filament profile based upon the middle range of the range of the printing temperature vs a typical ambient temperature (say 22.5C).  This would make sense one the Advanced page of the filament definition.

rtyr, 2021-02-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3059. Closing.

closed 2021-02-04 by rtyr