prusa3d/PrusaSlicer #3263: Scaling one dimension scales other dimensions

Frank0966, 2019-12-02:
### Version
_Version of PrusaSlicer used goes here_2.0.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)Prusa Mk3S
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_When trying to scale either X, Y or Z with lock off, the other two dimensions also change
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

JohnOCFII, 2019-12-02:
Just as a data point, I've been able to successfully scale a single dimension with PrusaSlicer 2.1.0+

Frank0966, 2019-12-02:
I'll wait for the update then. For now I can just scale it in 3D Builder
before going to PrusaSlicer

neophyl, 2019-12-02:
Prusa Slicer 2.1.0 is out and has been since mid september - https://github.com/prusa3d/PrusaSlicer/releases

Frank0966, 2019-12-02:
Thanks, I just downloaded it

Promite2501, 2019-12-02:
Works fine for me, there does not seem to be an issue here.  
Have you tried unlocking it the XYZ so it does not uniform scale?
[image]

bubnikv, 2020-03-15:
no response,implemented long time ago. closing.

closed 2020-03-15 by bubnikv