prusa3d/PrusaSlicer #3264: Supports are placed in air

JimmySnails, 2019-12-02:
### Version
_Version of PrusaSlicer used goes here_
2.1.0

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3S

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

When slicing the attached object with the 0.2mm quality profile, support is placed into thin air (see screenshots). Which seems to work but leads to spaghetti. i almost canceled my print before i realized that this is just support.

Is this intended? (Because it still works and it's only support) or is this a bug in support caluclation?

[image]

[Roof Middle Front.zip](https://github.com/prusa3d/PrusaSlicer/files/3912264/Roof.Middle.Front.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

The project also contains a second object (which has nothing to do with the issue)
[PrusaSlicer Project.zip](https://github.com/prusa3d/PrusaSlicer/files/3912278/PrusaSlicer.Project.zip)

bubnikv, 2021-02-24:
Duplicate of Supports that rest on an angled surface of the part don't work correctly (sometimes)
#4998

SimplyLiz, 2021-02-24:
@bubnikv If both issues are duplicates of each other, what's the solution here?

HiLivin, 2021-09-19:
The issue still persists in 2.4.0-alpha1 with either "Grid" or "Snug" support style on.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]