prusa3d/PrusaSlicer #3266: Solid bridge

espr14, 2019-12-03:
v2.1.0 win64

Bridge threads are printed next to each other. Due to lack of support they bend and change shape according to speed of cooling. But if you print them partially overlapped they will merge. This looks much better and the bridge is solid, not just couple of threads.

The only parameter that needs to be added to Slicer is amount of overlap.

Following example is 25% overlap with default 0.95 extrusion.

[gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/3915769/gcodes.zip)

[image]
[image]

supermerill, 2019-12-17:
Seems nice.
How did you test it?

espr14, 2019-12-17:
Prusament Galaxy black, Prusa i3 MK3S, default 0.4 nozzle, default settings. Manually modified G code so lines are 0.1 mm closer to each other without adjusting extrusion.

supermerill, 2019-12-17:
It's printed next to each other or you printed even bridges first then odds one?

ie ->'||||||' or ->'| | | ' then ->' | | |'

Tupson444, 2026-09-13:
still relevant