prusa3d/PrusaSlicer #3268: Better "Arrange" of objects on the plater.

Valthol, 2019-12-03:
I am printing a lot of different shaped objects, like 200 different shapes in my catalogue and I always want to fill my plater with them to the very full. PrusaSlicer "Arrange" function doesn't work well - it leaves a lot of space that when I manually squish objects  closer I get more.

It would be great if the program mathematically could do this job better than me, average human, and squish objects to the maximum and also there could be an additional option - which could allow the program to freely rotate objects (Z axis) so they could fill the plater even better.

This option would be AWESOME and I think it's relatively easy to implement. 

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-12-04:
> I think it's relatively easy to implement.

Yeah, people always do. @tamasmeszaros is the one who knows most about arrange function and I bet he's aware of room for improvement. Maybe there is an issue open already.

tamasmeszaros, 2019-12-06:
Hello @Valthol 
Thanks for the feedback, we always appreciate that. I'm sorry if the feature is stupid or frustrating. Unfortunately, the whole nesting an packaging problem is somewhat complicated and people do dissertations and stuff on this topic. Allowing rotations was discussed and it will be enabled in some future release when we will have time to get back to it and test it properly. You are right in that its not even that complicated as its already a feature of the nesting backend. 

However, this would be a predefined set of angles to try, not a completely free rotation. It would take ages to search a free interval of angles for each object.

The other problem is that the algorithm is working only for convex shapes and does not see (yet) the concavities and holes of the object silhouettes. Ones this gets solved *robustly and efficiently* (by someone somewhere in the scientific community or in-house), nesting will be radically more space-efficient.

The third problem is the space between objects to leave. The arrange should account for the support structures which are not generated when you hit the arrange button so a heuristic have to be used. Of course, it would be a good thing for this minimum distance to be available to the power user, but most users don't need to thinker with it.

The good thing is that if you feel something is missing and is easy to do, you can take matters into your own hands and try to do it yourself then do a pull request of your code and we will be happy to review and potentially integrate it into PrusaSlicer.

vmario89, 2019-12-22:
Hi, as also mentioned in https://github.com/prusa3d/PrusaSlicer/issues/1912, i appreciate to see some more intelligent algorithm like Rhoban's Plater, which is a really good working piece of software and at the moment the only good one in sense of open source

sslupsky, 2020-08-26:
@tamasmeszaros I think the rotations improvement as you suggested with predefined angles would be helpful.  Here is an example of a platter I recent customized where I rotate the part 180 degrees and overlap them so it minimizes the area on the platter.

Also, could there be a user adjustable setting for the distance between objects?

I noticed that you closed #1912.  What is the status of the "extension of the Slic3r nesting module with part-in-part placement" element of the roadmap?  When do you think we can expect to see something?

[image]

tamasmeszaros, 2020-11-20:
Hello @sslupsky ! The rotations and the adjustable distance is on the way, it may even fit into the next beta release. The part-in-part placement is still a plan, not receiving green light due to priorities of other features. At this point, integrating another library would be just as much work.

tamasmeszaros, 2020-12-08:
Hi @sslupsky ! Check out the updated arrangement with rotations and adjustable distance.

sslupsky, 2020-12-08:
Okie dokie ... this is in the 2.3 beta?

rtyr, 2021-02-04:
Implemented in 2.3.0. Closing.

closed 2021-02-04 by rtyr