prusa3d/PrusaSlicer #3273: "Single extruder multi material" option

smurfix, 2019-12-04:
### Version
2.any, current master

### 3D printer brand / version + firmware version (if known)
Prusa + MMS

### Behavior
Even with MM option on, there are five separate extruders in the dialog which need to be kept in mostly-identical state. Changing the nozzle diameter requires editing in five different places and clicking away the "must be identical" warning.

This is broken. Simple solution: Set the default values of extruders 2…5 to extruder 1 when not locked; disable changing the nozzle diameters of extruders 2+ when MM is on.

bubnikv, 2020-02-05:
It could be improved, but is it worth the added complexity?

smurfix, 2020-02-05:
Well, some people change nozzle sizes more often than others.

And, frankly, it's annoying. You need to click all those warnings away, and the program bloody well knows that there is a single nozzle, thus it should use that knowledge – instead of annoying the user.

The fact that each of these warnings neds to be clicked away explicitly is no help either.

Spiral Vase mode is much better, it tells you which settings are incompatible and lets you change them. One click, problem(s) solved.

lukasmatena, 2024-02-18:
This is solved since 2.6.0. The "must be identical" warning now offers to change the other values automatically. Closing.

closed 2024-02-18 by lukasmatena