prusa3d/PrusaSlicer #3275: Consistent crashes when trying to adjust support material on AMF re-import

Patronics, 2019-12-04:
### Version
2.1.0+

### Operating system type + version
_What OS are you using, and state any version #s_
MacOS Mojave, 10.14.6
### 3D printer brand / version + firmware version (if known)
Prusa Mk3S mmu2s

### Behavior
whenever I reopen the AMF file saved earlier in PrusaSlicer, and right click the part, and click "support material", PrusaSlicer crashes instantly, with this crash report generated: `Process:               PrusaSlicer [14193]
Path:                  /Volumes/VOLUME/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.1.0+-201909160907 (PrusaSlicer-2.1.0+-201909160907)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [14193]
User ID:               501

Date/Time:             2019-12-04 12:08:51.836 -0800
OS Version:            Mac OS X 10.14.6 (18G1012)
Report Version:        12
Bridge OS Version:     4.1 (17P1081)
Anonymous UUID:        5F41A753-FC4B-16C4-76F8-687E6C8D8148

Sleep/Wake UUID:       8F4AF0BB-A070-4030-9087-18949AEF42DA

Time Awake Since Boot: 230000 seconds
Time Since Wake:       1500 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000080
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [14193]

VM Regions Near 0x80:
--> 
    __TEXT                 000000010d065000-000000010df8b000 [ 15.1M] r-x/rwx SM=COW  /var/folders/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer

Application Specific Information:
Performing @selector(clickedAction:) from sender wxNSMenuItem 0x600001c289a0

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.prusa3d.slic3r/           	0x000000010d60146f Slic3r::GUI::ObjectList::get_freq_settings_choice(wxString const&) + 543
1   com.prusa3d.slic3r/           	0x000000010d610f20 std::__1::__function::__func<Slic3r::GUI::ObjectList::create_freq_settings_popupmenu(wxMenu*, bool)::$_33, std::__1::allocator<Slic3r::GUI::ObjectList::create_freq_settings_popupmenu(wxMenu*, bool)::$_33>, void (wxCommandEvent&)>::operator()(wxCommandEvent&) + 48
2   com.prusa3d.slic3r/           	0x000000010da30742 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 210
3   com.prusa3d.slic3r/           	0x000000010da3049b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
4   com.prusa3d.slic3r/           	0x000000010da303b7 wxEvtHandler::ProcessEvent(wxEvent&) + 103
5   com.prusa3d.slic3r/           	0x000000010da307ef wxEvtHandler::SafelyProcessEvent(wxEvent&) + 15
6   com.prusa3d.slic3r/           	0x000000010d8f6062 wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) + 82
7   com.prusa3d.slic3r/           	0x000000010d8f5f14 wxMenuBase::SendEvent(int, int) + 196
8   com.prusa3d.slic3r/           	0x000000010d806a75 wxMenu::HandleCommandProcess(wxMenuItem*, wxWindow*) + 117
9   com.apple.AppKit              	0x00007fff3a469644 -[NSApplication(NSResponder) sendAction:to:from:] + 312
10  com.apple.AppKit              	0x00007fff3a4c586b -[NSMenuItem _corePerformAction] + 323
11  com.apple.AppKit              	0x00007fff3a4c55dc -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 114
12  com.apple.AppKit              	0x00007fff3a52958f -[NSMenu performActionForItemAtIndex:] + 131
13  com.apple.AppKit              	0x00007fff3a5294fc -[NSMenu _internalPerformActionForItemAtIndex:] + 93
14  com.apple.AppKit              	0x00007fff3a52932e -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 105
15  com.apple.AppKit              	0x00007fff3a493dc6 NSSLMMenuEventHandler + 909
16  com.apple.HIToolbox           	0x00007fff3bddd22e DispatchEventToHandlers(EventTargetRec*, OpaqueEventRef*, HandlerCallRec*) + 1422
17  com.apple.HIToolbox           	0x00007fff3bddc5df SendEventToEventTargetInternal(OpaqueEventRef*, OpaqueEventTargetRef*, HandlerCallRec*) + 371
[... 458 lines cut ...]
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E4641474D4C4C522D4E5644
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E4641474D4C4C522D4E5644
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x843), wl0: Mar 28 2019 19:57:32 version 9.137.9.0.32.6.33 FWID 01-c5e1774f
Bluetooth: Version 6.0.14d3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: USB2.0 Hub
USB Device: AXE027 PICAXE USB
USB Device: iBridge Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 41.2
`
* _Expected Results_
It allows me to edit the support material options
* _Actual Results_
PrusaSlicer Crashes instantly

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
[Not Printing Sucessfully-BeltMesh-6Teeth-Inline - Bearing Mount.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3923762/Not.Printing.Sucessfully-BeltMesh-6Teeth-Inline.-.Bearing.Mount.3mf.zip)

FidelCapo, 2019-12-09:
I think the file you attached isn't the one you wanted. The file is 3MF not AMF and contains no parts. Can you check the attached file? Thank you

Patronics, 2019-12-09:
Oops, I Misspoke, it was a 3MF file, not AMF, you're right. That is the correct file though, and I verified (repeatedly, including redownloading the uploaded copy) that it does contain the (corrupted I guess?) parts when I import it. Here's a screenshot of the parts after opening them in PrusaSlicer. [image]

I experimented some more, and get the same problem when I right click and select many of the other options in the right click menu. Specifically, "Delete" works, "Export as STL..." works, and "Scale to print volume" work. "Change type" does nothing (is that normally even an option for non-modifier parts?). All the options below that (Infill, Layers and Perimeters, Support material, and all the options within the Add settings submenu) all cause the immediate crash. 
[image]

I hope this information helps, thanks for looking at this issue

YuSanka, 2019-12-12:
Thanks for your report. This bug is fixed and will be a part of a Release 2.2

closed 2019-12-12 by YuSanka