prusa3d/PrusaSlicer #3278: Slicing error with large file

Delfairen, 2019-12-05:
### Version
PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Win 10 64 bit build 18362
--------------------------------------------------------------------------
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 17,124MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.41
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 339 lines cut ...]
WGL_EXT_swap_control
--------------------------------------------------------------------------

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
Unable to slice the file.
1. select supports everywhere
2. click slice now

[image]

[StormBreaker-Hammer-Head.zip](https://github.com/prusa3d/PrusaSlicer/files/3924947/StormBreaker-Hammer-Head.zip)

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-03-15:
the message says it all. You need more RAM to slice that model.
Closing.

Delfairen, 2020-03-15:
Please try with the file as the memory usage on the machine when this happened was under 10%

Delfairen, 2020-03-15:
I will test with the latest version of the slicer and if I still get the error I will try on the server which has 48gb of RAM I'm it.

bubnikv, 2020-03-15:
The gyroid infill is a memory hog. It would not likely finish on my laptop with 16 GB RAM either.
Switching to a grid infill made PrusaSlicer to churn your model happily and quickly.

Delfairen, 2020-03-15:
Tested with 2.1.1 on Mac and it kills the slicer without an error message.

Delfairen, 2020-03-15:
Good to know. Thanks.

closed 2020-03-15 by Delfairen

bubnikv, 2020-03-16:
It would be great to improve, but it is not straightforward. A straightforward solution is to revert to some simplified flat rectangle preview instead of the thick extrusion models.