prusa3d/PrusaSlicer #3279: Support for Marlin CancelObject Gcode (M486)

NovaViper, 2019-12-05:
Hi, I am wondering if PrusaSlicer could get support for the newly added M486 command that Marlin introduced in v2.0. I know this feature is implemented in a similar way in with the cancel objects octoprint plugin but it seems like that plugin hasn't been updated in a while. And with the coming support of Python 3, that plugin might fall into a state where it can no longer be used. Here's some more information about how marlin implements it.

https://reprap.org/wiki/G-code#M486:_Cancel_Object
https://github.com/MarlinFirmware/Marlin/pull/15590

dtk1985, 2019-12-07:
➕1️⃣

chrisqwertz, 2019-12-21:
If you activate "Print Settings -> label objects" you already get something like this in the generated gcode:
```
; stop printing object HAPPY.stl id:0 copy 0
; printing object HAPPY.stl id:0 copy 1
```
then you could write a tiny python script that inserts a newline `M486 Sn` after every `id:x copy y` and call via "Post-processing scripts" at the bottom of the "Print Settings" tab.

paukstelis, 2020-02-08:
This should do what you want:

```
#!/usr/bin/python3
import sys
import re
import os

startregex = re.compile("; printing object (.*) id:(\d+) copy (\d+)")
endregex = re.compile("; stop printing object (.*) id:(\d+) copy (\d+)")

sourcegcode=sys.argv[1]

objects = []
ids = {"id" : 0, "copy" : 0, "index" : 0}
oindex = 0

def _get_entry(oid,copy,oindex):
    for o in objects:
        if o["id"] == oid and o["copy"] == copy:
            return o["index"],oindex
    objects.append({"id" : oid, "copy" : copy, "index" : oindex})
    oindex += 1
    return objects[-1]["index"],oindex

with open(sourcegcode, 'r+') as fd:
    contents = fd.readlines()
    for index, line in enumerate(contents):
        startmatch = startregex.match(line)
        if startmatch:
           obj_index, oindex = _get_entry(startmatch.group(2),startmatch.group(3),oindex)
           contents.insert(index + 1, "M486 S{0}\n".format(obj_index))

        endmatch = endregex.match(line)
        if endmatch:
           contents.insert(index + 1, "M486 S-1\n")

    contents.insert(0,"M486 T{0}\n".format(len(objects)))
    fd.seek(0)
    fd.writelines(contents)
```

NovaViper, 2020-02-10:
@paukstelis Thanks though PrusaSlicer 2.2.0 alpha 4 throws an error like this:
`!!!!! Failed to process the custom G-code template between_objects_gcode
Parsing error at line 9. Expecting a legacy variable expansion format
sourcegcode=sys.argv[1]
                     ^
!!!!! End of an error report for the custom G-code template between_objects_gcode`

paukstelis, 2020-02-10:
You should save that script as a python file (m486.py) somewhere on your system. Then just put the path to that file on the post-processing script section.

paukstelis, 2020-02-10:
Also, make sure you turn on "Label Objects" in the Output section.

NovaViper, 2020-02-15:
@paukstelis Oh, so the script should be on the pi?

paukstelis, 2020-02-15:
No. It should be wherever you are using PrusaSlicer. When PrusaSlicer writes the gcode file it will use this post-processing script to add M486 commands. You must have flashed your printer with Marlin Firmware that has been compiled to include M486 commands.

NovaViper, 2020-02-15:
But problem is, the slicer itself keeps throwing errors, not the firmware. I already sent what the error was saying in my previous reply.

⁣Get BlueMail for Android ​

paukstelis, 2020-02-15:
It looks from your error that you have put something in between_objects_gcode section which is NOT where the path to the script goes. Delete everything you have there and put the path the the python script in the `Print Settings->Output Options->Post-processing Scripts`

NovaViper, 2020-02-16:
@paukstelis What's the path though? How do I find the path of the script?

paukstelis, 2020-02-16:
The path is the location on your drive where you saved the file. I'd suggest reading up on how to use post-processing scripts in Slic3r/PrusaSlicer. This will probably help: http://projects.ttlexceeded.com/3dprinting_prusaslicer_post-processing.html

NovaViper, 2020-02-18:
@paukstelis Ah ok, but I have one other problem though. I have prusaslicer setup where the data directory for it is on Dropbox instead of on the computer, so the paths of the files can change up between my desktop and laptop (My desktop in particular has all programs on the SSD but the user directory on the harddrive. How can I make it where it only reads where the folder is and not care about the assigned drive letter of the path?

your-friend-alice, 2022-01-07:
@paukstelis a couple tweaks to the script (tested to produce identical outputs).

- uses a dictionary to track object ids instead of a list (should improve performance for files with many objects)
- avoids loading the whole file into memory (should help with large files on low-memory machines; unfortunately this requires processing in 2 passes in order to insert the total object count at the beginning).

From my testing it produces identical output to your script so should be safe:

```
#!/usr/bin/python3
import os
import re
import sys

startRegex = re.compile("; printing object (.*) id:(\d+) copy (\d+)")
endRegex = re.compile("; stop printing object (.*) id:(\d+) copy (\d+)")

srcPath = sys.argv[1]
dstPath = f'.{srcPath}.tmp'

objects = {}

def _getIndex(oid, copy):
    key = (oid, copy)
    if key in objects:
        return objects[key]
    else:
        newIndex = len(objects)
        objects[key] = newIndex
        return newIndex

def _matchStart(line):
    match = startRegex.match(line)
    if match:
       return (match.group(2), match.group(3))

def _matchEnd(line):
    return bool(endRegex.match(line))

if __name__ == '__main__':
    with open(srcPath, 'r') as src:
        # first pass just to gather object ids, needed for the first M486 line
        for line in src:
            startmatch = _matchStart(line)
            if startmatch:
               _getIndex(*startmatch)
        src.seek(0)
        with open(dstPath, 'w') as dst:
            # second pass to actually process the gcode
            print("M486 T{0}".format(len(objects)), file=dst)
            for line in src:
                # first, pass through the original line
                print(line.rstrip('\r\n'), file=dst)
                # if this line marks an object start, add the equivalent M486 code
                startmatch = _matchStart(line)
                if startmatch:
                   print("M486 S{0}".format(_getIndex(*startmatch)), file=dst)

                # if this line marks an object end, add the equivalent M486 code
                if _matchEnd(line):
                   print("M486 S-1", file=dst)
    os.rename(dstPath, srcPath)
```

arekm, 2023-05-10:
Could this be just backported from SuperSlicer?

https://github.com/supermerill/SuperSlicer/commit/b4e4b09a035eb35c1521b14e780b6eb94ba862de

Gona2119, 2023-07-07:
> @paukstelis a couple tweaks to the script (tested to produce identical outputs).
>
> * uses a dictionary to track object ids instead of a list (should improve performance for files with many objects)
> * [...]

Got an error in line 39, says "Line 39, in <module> with open(dstPatch, 'w') as dst:
OSError: [Errno 22] Invalid argument '.C:\\Users\\user\\Downloads\\CLIP.gcode.tmp'

your-friend-alice, 2023-07-09:
@Gona2119 ah, i failed to take into account that this could be called with an absolute path. the `dstPath` variable on line 10 is just `srcPath` with a `.` prepended and a `.tmp` appended, but on windows this'll obviously fail if the path starts with a drive letter, and on linux it'll fail more subtly by falsely turning absolute paths into relative paths. I've added a `_tmpName` function that properly splits up the path before adding the leading dot, this should fix this

```
#!/usr/bin/python3
import os
import re
import sys

startRegex = re.compile("; printing object (.*) id:(\d+) copy (\d+)")
endRegex = re.compile("; stop printing object (.*) id:(\d+) copy (\d+)")

def _tmpName(path):
    head, tail = os.path.split(path)
    tail = os.path.basename(path)
    return os.path.join(head, f'.{tail}.tmp')

srcPath = sys.argv[1]
dstPath = _tmpName(srcPath)

objects = {}

def _getIndex(oid, copy):
    key = (oid, copy)
    if key in objects:
        return objects[key]
    else:
        newIndex = len(objects)
        objects[key] = newIndex
        return newIndex

def _matchStart(line):
    match = startRegex.match(line)
    if match:
       return (match.group(2), match.group(3))

def _matchEnd(line):
    return bool(endRegex.match(line))

if __name__ == '__main__':
    with open(srcPath, 'r') as src:
        # first pass just to gather object ids, needed for the first M486 line
        for line in src:
            startmatch = _matchStart(line)
            if startmatch:
               _getIndex(*startmatch)
        src.seek(0)
        with open(dstPath, 'w') as dst:
            # second pass to actually process the gcode
            print("M486 T{0}".format(len(objects)), file=dst)
            for line in src:
                # first, pass through the original line
                print(line.rstrip('\r\n'), file=dst)
                # if this line marks an object start, add the equivalent M486 code
                startmatch = _matchStart(line)
                if startmatch:
                   print("M486 S{0}".format(_getIndex(*startmatch)), file=dst)

                # if this line marks an object end, add the equivalent M486 code
                if _matchEnd(line):
                   print("M486 S-1", file=dst)
    os.rename(dstPath, srcPath)
```

lukasmatena, 2024-02-06:
The M486 support is implemented since 2.7.0. Closing.

closed 2024-02-06 by lukasmatena