prusa3d/PrusaSlicer #3281: Feature Request: Block the use of % in the filename when saving gcode

jweaver100, 2019-12-05:
### Version
2.1.0+win64

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Prusa lets you use "%" in the filename when you export/save gcode.. And this might not seem like a problem and certainly isn't under Windows.

But i use a FlashAir card, and if you create a file with a "%" it gets into state where you can no longer write to it... And you can't delete the file. The file actually appears called something like this:

"http://192.168.1.120/filename%.gcode"

You can't delete this file via the NFS mount, or the web interface. The only way to fix  is to take the card out, mount it on a PC and delete the file.. Not a big deal, but annoying.

(I think its the /s that makes this file undeletable rather than the %, but its the % which triggers this)

I often forget this and if I reduce or enlarge something I name the file something like ".... at 60%" and after this the Flashair card is unusable until i can remember how to fix it.

So could you consider blocking the use of % when you save a file?

neophyl, 2019-12-05:
Its not just the % that the flashair has trouble with iirc, its other  non alphanumeric special characters too.  I think its because they are used by webdav (so similar to html) where to use they have to be escaped.  
I have had it happen a couple of times to me too but its an annoyance only.  I'm sure if the devs limited the available characters they would have many more complaints about such limiting and the change could break other peoples existing workflows and naming systems.

jweaver100, 2019-12-05:
Makes sense.. What about a feature (option) where you can choose between a full or reduced character set.

So by default it is as it is today? And there is an option which can be used to change to a basic alpha-numeric set?

closed 2026-09-29 by prusaslicer-bot[bot]