prusa3d/PrusaSlicer #3283: Multi-extruder complete-single-object mode drops filament assignments

smurfix, 2019-12-05:
### Version
2.x, current master

### Operating system type + version
Linux testing

### 3D printer brand / version + firmware version (if known)
Prusa MM

### Behavior
* Rename the attached file to something.3mf and load it
* Click "Slice now"
* A wipe tower appears, objects have the correct tool/extruder color
* Move the mouse onto the plater. The wipe tower disappears, suddenly all objects use extruder 4. Oops.
* Export to gcode. Verify with an editor that this is not just a visual problem, the resulting gcode only contain T3 commands but no T0.

#### Project File (.3MF) where problem occurs

[whole_model_problem.zip](https://github.com/prusa3d/PrusaSlicer/files/3927702/whole_model_problem.zip)

lukasmatena, 2019-12-05:
You are printing sequentially (Print Settings->Output Options). The wipe tower is not supported for sequential prints. The only real bug is that there is no error message. We should add it.

smurfix, 2019-12-05:
OK, so the problem should go away if I disable the wipe tower … testing … thanks, that seems to have worked. In any case, why no wiping for sequential prints? the material from the previous filament still needs to go somewhere.

I just found another potential problem with sequential printing, in that while the initial bed temperature is re-set, he printer doesn't wait for it. Is that intentional?

lukasmatena, 2019-12-06:
> why no wiping for sequential prints? 

Because it is not as simple as it probably looks. Let's say you have several multimaterial objects that you want to print sequentially. In that case, you'd need a wipe tower for each of them, which complicates everything and is not going to be used often anyway, because of long print time and material cost. It would also be possible to print one wipe tower and always travel to its current height, but then the wipe tower can grow higher than the print area, which is another corner case and there may be others. Probably the only case that makes some sense is the one demonstrated by you, that is, several single color objects. This makes it a low-priority task with high risk of breaking something that's been already tested.

> while the initial bed temperature is re-set, he printer doesn't wait for it. Is that intentional?

It is. The idea is that because the temp change happens before the z-travel, there should be plenty of time to reach the temperature. Did you have some actual issues with the behaviour?

smurfix, 2019-12-06:
> Let's say you have several multimaterial objects that you want to print sequentially.

Point. However, my usecase has five different-mono-filament objects. That'd require a wipe square, not a wipe tower. ;-)

> The idea is that because the temp change happens before the z-travel, there should be plenty of time to reach the temperature.

I don't know about you but my printer takes 5sec per degC to warm up. Small objects (I'm doing a bunch of covers for unused lamp outlets) are on layer two before the target is reached.

lukasmatena, 2019-12-11:
The error message when one attempts to use the wipe tower in sequential printing mode was added with https://github.com/prusa3d/PrusaSlicer/commit/e7475772713fd1cdefee066d5fd06194f329cfad to be part of PrusaSlicer2.2.0.

closed 2026-09-29 by prusaslicer-bot[bot]