prusa3d/PrusaSlicer #3286: new slicer keeps crashing, windows 10 64bit

stanzy75, 2019-12-06:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-12-06:
Please, provide more detailed info (or _some_ info at least). We need to know exact version of PrusaSlicer (or git revision if you compiled it yourself) and ideally see a log. Run PS from command line as
`prusa-slicer-console.exe --loglevel=9`. The application will then log to the console.

You can also try running `prusa-slicer-console.exe --loglevel=9 --sw-renderer` to disable hardware acceleration to see if that's not the cause.

bubnikv, 2020-03-15:
no response, closing.

closed 2020-03-15 by bubnikv