prusa3d/PrusaSlicer #3290: Recommended object thin wall thickness not using customized default extrusion width

brnnrc, 2019-12-07:
### Version
2.1.0
### Operating system type + version
Win64
### Behavior
Changing the advanced default extrusion width does not affect the "Recommended object thin wall thickness" calculation. Otherwise changing perimeters and external perimeters widths calculates as expected.

bubnikv, 2020-03-15:
please provide your config or a 3MF.

greendog99, 2020-06-20:
I've struggled with this as well and came here today to see if anyone else had reported it yet. I've narrowed down the problem, hopefully this will help:

## Version
2.2.0+

## Operating System
MacOS 10.15.5

## Steps to Reproduce
1. Set "Advanced" -> "Default extrusion width" to `0.5`
2. Set "Advanced" -> "Perimeters" to `0`
3. Set "Advanced" -> "External Perimeters" to `0`
4. "Layers and Perimeters" page reports `Recommended object thin wall thickness for layer height 0.20 and 2 lines: 0.86mm`. This is **incorrect**.

The help text for "Perimeters" and "External Perimeters" reads: *"If left zero, default extrusion width will be used if set"*. However, this does not seem to be happening.

When the default extrusion width is changed and the two perimeter widths are set to zero, the walls correctly print at the new extrusion width, but the recommended thin wall thickness does not seem to use the new default extrusion width in its calculations.

If the two perimeter width parameters are manually set to `0.5`, then the recommended object thin wall thickness is reported correctly as `0.96mm`.

Screenshots:

Change settings:
[image]

Thin wall thickness reported incorrectly:
[image]

Manually set perimeter width to match default width:
[image]

Thin wall thickness is now correct:
[image]

Perimeter width help text:
[image]

Thank you!

closed 2026-09-29 by prusaslicer-bot[bot]