prusa3d/PrusaSlicer #3291: Places for change of filament

espr14, 2019-12-07:
v2.1.0 win64

When filament runs out it pauses the print and waits. This can happen anywhere during print including outside perimeter. Such change is visible which is not good.

Slicer could insert areas (starts and stops) where filament can be changed without impact on the look of printed result. Such areas could be infill, wipe tower, inside perimeters, inside solid infills. And definitely not the outside perimeters or bridges. I suppose there's no G-code yet but could discussion can be initiated.

BeldrothTheGold, 2019-12-12:
Are you talking about the filament runnout sensor triggering a pause on the MK3? Or are you talking about where manual color change pauses happen when they are inserted using PrusaSlicer?

espr14, 2019-12-12:
Color change is defined during slicing and this happens after the whole layer is done which is OK.

I'm talking about filament runout which can happen anywhere and could make a mess on the external perimeter. If the sensor is moved away from the print head, there could be enough time to print whole external perimeter and change the filament during printing something less visible.

BeldrothTheGold, 2019-12-12:
Neither PrusaSlicer nor the Firmware knows how much filament is left on the spool. So it has no way to plan for a filament change triggered by the runout sensor.

I don't know if there is any safe way to allow the printer to continue to plan moves to a "change area" after the runout is triggered either. Not without risking running the remaining filament past the end of the extruder gears so it is unable to back the filament out for you to change it.

BeldrothTheGold, 2019-12-12:
I apologize. I missed where you said to move the sensor away from the extruder. 

Yeah, I suppose it would be possible if your still using a direct drive extruded to do this. Would require coordination between the firmware and Slicer dev teams to implement.

Have the Firmware read ahead for a notification of a Safe Zone or whatever you want to call it. Something added to the gcode by the Slicer.

espr14, 2019-12-12:
Imagine something like i3 MK3 with MMU2 (MMU3). You can have sensors at the spools so you can have like 200 mm of filament available before reaching point of no return. That means uninterrupted print of 10 kg of filament (5x 2 kg).

Heckw, 2019-12-16:
How can you be sure that all people with MK3(s)+MMU2 use filament sensors at the spools?
My 1st filament sensor is the finda sensor in the MMU. This leaves about 40cm of filament until the nozzle goes out of filament.

espr14, 2019-12-16:
It's up to them where to put filament sensor. The actual distance should be handled by firmware of the printer. It can be defined from manufacturer or changed by user. This feature is an option. If the firmware does not understand it, it will just skip it.

closed 2026-09-29 by prusaslicer-bot[bot]