prusa3d/PrusaSlicer #3295: Outer Perimeter Underextrusion

Petrda8, 2019-12-08:
### Version
_PrusaSlicer-2.1.0+-201909160907_

### Operating system type + version
_MacOS 10.15.1 (19B88)_

### 3D printer brand / version + firmware version (if known)
_Prusa i3 Mk3s, purchased Nov 2019_

### Behavior
I'm having under-extrusion issues on all my prints (with varying impacts, from aesthetical to functional). I found this [thread](https://github.com/prusa3d/PrusaSlicer/issues/806) with another user having the same issues back in 2018. I'm using Prusa Slic3r with default settings for my printer, also Prusament grey PLA that came with the printer. The rest of the print is awesome, no issues whatsoever, but this issue is very annoying. 

Here are a couple of pictures, not all of them are related to this issue but most of them:
https://imgur.com/a/7CrsIoJ (very clear example)
https://imgur.com/a/V1jxjMR (e.g. the panther or IrisBox)

Many thanks in advance!

_Is this a new feature request?_ No

matlinder, 2019-12-15:
following this thread, I am having the same issues, the layer start causes under extrusion

matlinder, 2019-12-18:
Try this profile, it solved my issues and made my prints even better

[Pretty PLA v4](https://prusaslicerprofiles.com/download/pretty-pla-v4-0/)

Petrda8, 2019-12-20:
Thanks Mat, will give it a try! :-) 
In any case, would be great to reflect that within Slic3r directly as I'm sure other guys might have similar issues.

matlinder, 2019-12-25:
Well it made my prints nicer, but the issue still persists. Sometimes it is not even at the seam, it is just random.

Petrda8, 2019-12-30:
Still happening as well :( this one is nicely hidden on the inner side, but nevertheless still there https://imgur.com/a/tHYHf8r

Hope Prusa Team gets to this one to have a look

stelgenhof, 2020-01-13:
I'm experiencing a similar - or even the same - issue. See the below examples:

[image]
[image]

The under extrusions are at the locations of the start and/or end of a layer. For the orange part, I tried to reduce the acceleration.  My assumption was that, based on the way it looked like, the acceleration was too high so that not enough filament was deposited at the corners. However reducing the acceleration didn't help.

I got the issue 'solved'/minimized by re-orientating the part, changing where the same is located, and printing with outer perimeters first. See the same orange part below after experimenting with different settings/orientation and also retraction.

[image]

There are still little holes/gaps at some points where a line starts or ends. Therefor I believe the issue is not solved.

**Printer**: MK2.5
**Filament**: PLA (Silver) / ABS (Orange) - All calibrated and checked
**Slicer**: PrusaSlicer - standard profile (0.2mm)

I remember not having this issue before recently (i.e. in the last 6 months)...

Cheers! Sacha

vintagepc, 2020-11-03:
I am also observing this with 2.3.0-alpha2, whereas previously I would not see this issue. 

It does not appear to be a K-value issue as not all deretractions on a layer are affected, only one of them is.

rtyr, 2023-04-04:
No response. Closing.

closed 2023-04-04 by rtyr