prusa3d/PrusaSlicer #3300: Random value retractions added

tonycstech, 2019-12-10:
### Version
2.1.0+win64
### Operating system type + version
Windows 10 X64

### Behavior
Code line 71 After completing skirt print, desired retraction of -2mm takes place.
Right after -2mm, another -20.57mm retraction is added for no reason.
Primes only by 2mm, does not prime 20.57mm
Code line 117 retracts -2mm and then another 0.79mm retraction is added for no reason.

[Front.zip](https://github.com/prusa3d/PrusaSlicer/files/3944229/Front.zip)

lukasmatena, 2019-12-10:
There is `G92 E0` command in between, which resets extruder axis to zero. So what is really happening should be this:

```
G1 X2.894 Y1.233 E22.57240   // last extrusion move
G1 E20.57240 F2400.00000     // retract 2 mm
G92 E0                       // reset extruder axis to zero
G1 Z0.630 F9000.000          // z-hop
G1 X32.671 Y181.788          // travel
G1 Z0.330                    // undo the z-hop
G1 E2.00000 F2400.00000      // unretract 2mm
G1 F1800                     // set feedrate
G1 X33.331 Y181.904 E2.02384 // next extrusion move (0.02384)
```

Did you have any actual issues printing it? What printer, what firmware?

tonycstech, 2019-12-10:
Printer or firmware does not matter.
I preview the code before I print using craftware, best print preview in
the world.
I won't even try to print it when I see things like that. Why waste time
and filament.

lukasmatena, 2019-12-10:
I'm sorry, but we are trying to produce gcodes that print correctly, not ones that correctly display in Craftware, the best software in the world.

tonycstech, 2019-12-10:
It generates unwanted retraction.
It will not print correctly like that.

Am sorry, I must have posted my finding in the forest for trees to look at.

lukasmatena, 2020-01-09:
I'm closing the issue, there is no problem with the gcode as far as I can tell and there is also no report of a failed print. How it shows is Craftware is Craftware's problem.

If anyone actually printing with an absolutely addressed extruder tells me there is a problem, it can be reopened.

closed 2020-01-09 by lukasmatena

tonycstech, 2020-01-09:
It was my fault. Problem was Craftware. For what ever reason Prusa generated code was making it think that there were retractions when in fact there never was.
I ran the print and it had no random retractions. 
My bad.
I typically look at the preview before i try to print. To this day Craftware was very reliable and best so far because it was actually showing me where nozzle is and whats its doing.
I wish how ever Prusa had that feature: Show virtual nozzle and display length extruded/retracted.

lukasmatena, 2020-01-09:
Ok, thanks for the reply. Gcode preview improvements along those lines is in the planning, but I better won't promise when it will be ready. It is quite a lot of work.