prusa3d/PrusaSlicer #3301: Music tones are broken in version 3_8_1_2869

WhiteDragonD, 2019-12-11:
### Version
_Version of PrusaSlicer used goes here_
2.1.0

### Operating system type + version
Windows 7/10

### 3D printer brand / version + firmware version (if known)
MK2.5s

### Behavior
In Firmware version 3.5.1-1778, I could add the music tones at the end of a print (after M84 ; disable motors) and then it would play the music tones.  With firmware version 3_8_1_2869, it just plays beeps without changing the tones.  

Example of music tones inserted after M84:

;MarioOverworld
M300 S0 P75
M300 S2637 P150
M300 S2637 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2637 P150
M300 S0 P150
M300 S3135 P150
M300 S0 P450
M300 S1567 P150
M300 S0 P450
M300 S2093 P150
M300 S0 P300
M300 S1567 P150
M300 S0 P300
M300 S1318 P150
M300 S0 P300
M300 S880 P150
M300 S0 P150
M300 S987 P150
M300 S0 P150
M300 S932 P150
M300 S880 P150
M300 S0 P150
M300 S1567 P150
M300 S2637 P150
M300 S3135 P150
M300 S1760 P150
M300 S0 P150
M300 S2793 P150
M300 S3135 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2349 P150
M300 S987 P150
M300 S0 P600
M300 S3135 P150
M300 S2959 P150
M300 S2793 P150
M300 S2489 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S1661 P150
M300 S880 P150
M300 S2093 P150
M300 S0 P150
M300 S880 P150
M300 S2093 P150
M300 S2349 P150
M300 S0 P300
M300 S2489 P300
M300 S0 P150
M300 S2349 P150
M300 S0 P300
M300 S2093 P300;
M73 P100 R0

#### Project File (.3MF) where problem occurs

WhiteDragonD, 2019-12-11:
I have now confirmed that version 3_7_2_2363 works but version 3_8_0_2684 on up does not.  Please fix.

BeldrothTheGold, 2019-12-11:
I think this issue was created in the wrong repository. If it is a firmware issue you submit the issue on this repo: https://github.com/prusa3d/Prusa-Firmware.

closed 2019-12-11 by lukasmatena