prusa3d/PrusaSlicer #3304: Extra z hops only in the 1st layer

espr14, 2019-12-11:
v2.1.0 win64

I set:
- lift z hop to 2 mm, 
- Avoid crossing perimeters to on,
- Only retract when crossing perimeters to on

This works great but it creates extra z hops in the first layer only. You can see it in the g code. Since the object is in one piece from bottom to top, there's no retraction needed so no z hops should occur.

[stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3949679/stl.zip)
[stl.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3949680/stl.3mf.zip)

supermerill, 2019-12-17:
So maybe a setting "no z-hop for the first layer" ?

bubnikv, 2020-12-11:
More details, for example pictures with markings would be welcome.

espr14, 2020-12-12:
I've revisited this issue in PS 2.3.0 b2. There are some layers where crossing of perimeters are inevitable. But the rest should be without retraction. These are all cases with solid layer but retraction is triggered (and should not be (with exception of skirt of course)):
[image]
[image]
[image]
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]