prusa3d/PrusaSlicer #3305: Double path thin wall

espr14, 2019-12-11:
v2.1.0 win64

I have thin external perimeter due to thin wall. It's a cylinder extracted from a cube. Normally, it won't create a solid wall at all - not good. If I decrease extrusion width, it will finally connect the wall and make two external perimeters.

So in case of thin external perimeter, slicer could decrease extrusion width for the area so two externals perimeters are printed. This looks much better and stronger while preserving fast print for the rest of the model (keeping default extrusion width). User could set what's the minimal extrusion width allowed (e.g. 40%).

Default extrusion width:
[image]

Decreased extrusion width:
[image]

supermerill, 2019-12-17:
You can't print a thinner extrusion than your nozzle width * 1.05

You can set your external perimeter at the finest possible (should be the default setting) and thicker for the other ones. The gap problem the first picture is due to the bad "thin wall" algorithm, imo.

Also, be careful with % in width settings, it's a % of the layer height.

espr14, 2019-12-17:
You can print line thinner than your nozzle diameter. The issue is quality only. But that's the same for thicker lines.

closed 2026-09-29 by prusaslicer-bot[bot]