prusa3d/PrusaSlicer #3308: Crashes when I set 0 retraction

MJDien, 2019-12-12:
### Version  2.1.0+win64

### Operating system type + version
win10
### 3D printer brand / version + firmware version (if known)
Moddified Anet A8 marlin 1.1.x
### Behavior
Tried to set everything up to print with TPU. When I change the retraction to 0mm and try to slice the object Prusaslicer crashes. It works with re: 0.5 mm

lukasmatena, 2019-12-14:
Please enclose a 3MF.

Wannageering, 2020-01-04:
Same issue for me, all goes fine until the end of the calculations and then it crashes. Same model, same printer, the only setting changed is retractions = 0 mm

lukasmatena, 2020-01-06:
I now realized that this is most likely a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3124, which was already fixed. You can retest in PrusaSlicer2.2.0-alpha2.

I'm closing the issue. In case I'm wrong and this is something different, let us know and we can reopen it.

closed 2020-01-06 by lukasmatena