prusa3d/PrusaSlicer #3311: PrusaSlicer is not using correct layer height when calculating width/layer height for height range modifiers

napter, 2019-12-13:
### Version
2.1.0

### Operating system type + version
Windows 10 

### Behavior
When using a height range modifier to modify the layer height and the extrusion width, PrusaSlier throws an error that the extrusion width is too low to be printable for the layer height. It is incorrectly using the main layer height in this calculation and not the layer height from the height range modifier.

To reproduce:
Set layer height to 0.3 in layers and perimeters
Add a height range modifier
In the modifier set layer height to 0.25
In the modifier set any extrusion width to 0.3
PrusaSlicer will throw the error "external_perimeter_extrusion_width=0.3 mm is too low to be printable at a layer height of 0.3 mm. 

However, the 0.3 mm width is being printed at a layer height of 0.25 not 0.3. It is not using the modifier's layer height when doing the calculation for this error.

DRKMSTR, 2020-10-16:
Bump

This issue still has not been fixed. I cannot increase layer height to anything higher than 0.25mm

DRKMSTR, 2020-10-16:
Found the issue: Go to the "Printer Settings" Tab, select "Extruder 1" and set "Layer Height Limits" to "0.3mm" or higher.

bubnikv, 2021-01-20:
So the issue is solved, right?
If so, please close it.

napter, 2021-01-21:
I don't think whatever DRKMSTR was talking about had anything to do with the issue I described. He had his own issue and then resolved it.

rtyr, 2021-02-04:
I can still reproduce it in 2.3.0.

ubqtous, 2022-06-05:
Ran into this issue today and found this post from Google. Its still not fixed in 2.4.2. Hello??

closed 2026-09-29 by prusaslicer-bot[bot]