prusa3d/PrusaSlicer #3313: Multi extruder setup: No tool selection emitted when using the first tool (Extruder 1)

AlessioMorale, 2019-12-13:
### Version
version 2.1.1

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Custom printer/Marlin 2.0.0 / Dual extruder setup.
### Behavior
When printing with Extruder 2 It emits `T1` to correctly select the extruder to use.
When using Extruder 1 no `T0` is emitted causint the printer to print with wathever was the last extruder used.
  
* In a multi extruder setup select Extruder 1 for the part in Plater tab
* Resulting Gcode has no `T0` so the printer will print using the last extruder selected by any previous command instead of Extruder 1.
* Gcode should contain `T0` to force the selection of Extruder 1
* No `T0` in gcode

Attached two gcodes for extruder 0 and 1

[gcode_extruder_0.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/3962836/gcode_extruder_0.gcode.txt)
[gcode_extruder 1.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/3962837/gcode_extruder.1.gcode.txt)

lukasmatena, 2019-12-14:
Could you please enclose a 3MF with the complete project so we can try to reproduce the problem? I tried with the config loaded from your gcode and all seemed fine. Thanks.

AlessioMorale, 2019-12-14:
Thanks for your reply. 
Attached two files with both settings (extruder 1 and 2) and the resulting gcodes.
[test_projects.zip](https://github.com/prusa3d/PrusaSlicer/files/3963774/test_projects.zip)

lukasmatena, 2019-12-16:
Ok, I understand now. The initial extruder is supposed to be loaded through custom start gcode. After it is processed, PrusaSlicer expects that the printer is ready to print and correct filament has been already loaded. You can have a look at our MMU2 profile. You essentially need to add at least `T[initial_tool]` in there.

closed 2026-09-29 by prusaslicer-bot[bot]