prusa3d/PrusaSlicer #3316: PrusaSlicer crashes with segmentation fault

mkbel, 2019-12-14:
### Version affected
PrusaSlicer-2.1.1+linux64-201912101511.AppImage
PrusaSlicer-2.1.0+linux64-201909160925.AppImage
### Version unaffected
PrusaSlicer-2.0.0+linux64-201905201652.AppImage

### Operating system type + version
Debian sid AMD64 with XFCE 4.12.5

### 3D printer brand / version + firmware version (if known)
unrelated

### Behavior
* _Steps needed to reproduce the problem_
  * Open recent project zebrik_vesak_crash.3mf
  * Click on Slice now button
* _Expected Results_
  * Project is sliced to gcode.
* _Actual Results_
`DRI_PRIME=1 ./PrusaSlicer-2.1.1+linux64-201912101511.AppImage 
** (prusa-slicer:25444): WARNING **: 19:12:29.885: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-OgSZRommSF: Connection refused
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.588: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.814: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.815: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.815: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.815: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.815: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.815: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.859: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.859: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.859: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.859: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.859: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:25444): Gtk-CRITICAL **: 19:12:31.859: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
Segmentation fault`

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
This file is not zipped, just rename the file to zebrik_vesak_crash.3mf .
[zebrik_vesak_crash.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3964159/zebrik_vesak_crash.3mf.zip)

lukasmatena, 2019-12-16:
I cannot reproduce it in current master, it seems to be already fixed for 2.2.0 release. The problem is most likely the layer range modifier that does not overlap the object at all - there was a bug related to it that I remember was fixed (6a07b231e19c976e86f186bdba0ff0b59a35bdc6).
I does indeed crash in 2.1.1.

lukasmatena, 2020-03-04:
Should be fixed, closing.

closed 2020-03-04 by lukasmatena

mkbel, 2020-03-04:
Confirmed that it doesn't crash on 2.2.0-alpha4.