prusa3d/PrusaSlicer #3317: Speed modifier cuts object in half

mkbel, 2019-12-14:
### Version
PrusaSlicer-2.1.1+linux64-201912101511.AppImage
PrusaSlicer-2.0.0+linux64-201905201652.AppImage

### Operating system type + version
Debian Sid AMD64 with XFCE 4.12.5

### 3D printer brand / version + firmware version (if known)
unrelated

### Behavior
* Speed modifier cuts object in half
* _Steps needed to reproduce the problem_
  * Add box modifier to part of object
  * Set perimeter/outer perimeter/small perimeter speed to non-default value
* _Expected Results_
  * Perimeter print speed is modified in modifier area (print slows down or speeds up in area)
  * There is no change in perimeter shape
[image]

* _Actual Results_
  * Perimeter print speed is modified in modifier area
  * Additional perimeters are generated as if the object were cut on border of modifier weakening the object and increasing print time.

[image]

_Is this a new feature request?_
No, I think it is bug.

#### Project File (.3MF) where problem occurs
This file is not zipped, just rename it to zebrik_vesak_MK3_laying_modifier_speed_bug.3mf .
[zebrik_vesak_MK3_laying_modifier_speed_bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3964413/zebrik_vesak_MK3_laying_modifier_speed_bug.3mf.zip)

supermerill, 2019-12-18:
That's how it's done internally: if the speed isn't the same, it separate the ares and slices them separately to be able to store them in separate containers, to be able to set the correct speed to each one. So it's not really a bug but a feature, as it's intentional.

Ideally, it should be sliced as one piece and the cut done at gcode generation. But nothing is in place to make it, as this feature priority may be very low.

BMacours, 2023-01-11:
As commented in #3370 : +1 for not "cuting" objects especially when modifying parameters which are not affecting perimeters (i.e. speed, ...). I'm printing small technical parts with overhangs where I have to reduce speed (to limit curling) without changing the part visual aspect nor it's strength caused by added breaking perimeters. Thanks!

nergen, 2023-11-01:
+1 to this issue since my was closed as duplicate

Shushuda, 2026-09-02:
This issue is still occurring in 3.0.0 alpha.