prusa3d/PrusaSlicer #3319: Centering modifiers with object manipulation 0mm does not work

mad2xlc, 2019-12-15:
### Version
2.1.0 + 2.1.1 win64

### Operating system type + version
Windows 10 64bit 18363

### Behavior
* _Describe the problem_
It is not possible to move a modifier, support enforcer or blocker to position 0mm in any direction (X, Y or Z) by using object manipulation fields on the right side.

* _Steps needed to reproduce the problem_
It is possible to enter 1mm and you get 1mm. But if you enter 0mm so it is center aligned to the main object, it does not work.
As a workaround you can enter 0.001mm. This will be rounded to 0mm.
* _Expected Results_
Enter 0mm for X, Y or Z center the modifier to the main object
* _Actual Results_
Input to these number fields are ignored if 0 is entered.
_Is this a new feature request?_

lukasmatena, 2019-12-15:
I think this is a duplicate of #3108 (and the others). It was already fixed on master.

rtyr, 2021-02-04:
Fixed. Closing.

closed 2021-02-04 by rtyr