prusa3d/PrusaSlicer #3320: Bridge perimeter not detected

wavexx, 2019-12-15:
### Version
version_2.2.0-alpha0-95-gae2dbcc46 (master as of 2019-12-15)

### Behavior
In the following test case:

[missed_bridging_perimeter.zip](https://github.com/prusa3d/PrusaSlicer/files/3965030/missed_bridging_perimeter.zip)

"Detect bridging perimeters" has been enabled, but there are two sections of the mesh whose perimeter is not correctly detected as such.

Layer 19 shows the two sections being printed as regular perimeters:

[image]

Other bridged perimeters seem to work correctly; look at a few layers above under the ceiling:

[image]

I assume this has to do to the fact that the perimeter is near (or at) the start of a cusp, since this is a rare occurrence in my experience.

espr14, 2020-01-06:
This is caused by thin perimeter which is bridge at the same time. I've got another example.
[no bridge.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4027980/no.bridge.3mf.zip)

espr14, 2020-01-07:
Solution can be: if the segment of the path is not supported and it is thin then remove it.

wavexx, 2020-01-10:
Attaching a simple test model, related openscad file to generate a torture test and 3mf file for easy repro:

[slope_test.zip](https://github.com/prusa3d/PrusaSlicer/files/4046027/slope_test.zip)

This is a series of suspended wedges with an increasing angle. Ideally, there should be at least a bridge perimeter on each wedge, but in practice you can see it's not the case:

[image]

Not only that, but test also uncovers that for angles close to 90' (the last wedges) bridge infill is used incorrectly, causing overextrusion:

[image]

bubnikv, 2020-01-10:
Bridges are only correctly oriented if there is enough of infill for the
ends of the bridges to be anchored to.

wavexx, 2020-01-10:
This is not an issue about orientation though. The bridge-infill seems to actually be oriented fine.

The main issue in this case is that the bridging perimeter is not recognized as such when it's on a thin edge. I can increase the side anchoring (side walls from 2mm to 5 - for example), but it doesn't change the result.

nordurljosahvida, 2020-05-20:
+1 for non-detected bridge. here is an example:

[spike_faceshield_bridging_perimeters_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4658232/spike_faceshield_bridging_perimeters_issue.zip)

pblocz, 2021-01-13:
I was going to test bridging using this model from thinginverse https://www.thingiverse.com/thing:476845 and run into this issue. The bridges were detected as overhangs and bridging performance was really poor when printing.

Using version 2.3.0

[image]

closed 2026-09-29 by prusaslicer-bot[bot]