prusa3d/PrusaSlicer #3323: I would like to request an additional PrusaSlicer supports-feature. I want to use the "supports only from the build-plate" and be able to add further supports by the existing possibilities (cylinder, slab etc.).

Heckw, 2019-12-16:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Heckw, 2019-12-19:
I would like to request an additional PrusaSlicer supports-feature. I want to use the "supports only from the build-plate" and be able to add further supports by the existing possibilities (cylinder, slab etc.).

bubnikv, 2020-03-15:
both features are there for a long time.

closed 2020-03-15 by bubnikv

Heckw, 2020-03-27:
I just checked it with 2.2.0 and it these functions DON'T function together. Please note that I want to combine these support features!
First I choose "supports only from the build-plate" AND THEN I want to add ADDITIONAL supports using the existing possibilities (cylinder, slab etc.) !!