prusa3d/PrusaSlicer #3326: Bad slicing in vase mode (unexpected bridge and infill layers)

ochm, 2019-12-16:
### Version
2.1.1+win64

### Operating system type + version
Win10 64bit CZ

### 3D printer brand / version + firmware version (if known)
i3 MK3S, 3.8.1-2869

### Behavior
After slicing object, from 17th to 21st layer is unexpected bridge and solid infill.
[image]
[image]

Checked the same in Cura where is sliced OK
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[lampa-crestas-c.zip](https://github.com/prusa3d/PrusaSlicer/files/3970293/lampa-crestas-c.zip)

supermerill, 2019-12-17:
Quick fix: add a modifier volume for the whole object but the first layers (create a big box and move it a bit in the z axis) and set "bottom perimeters" to 0 for this volume.

ochm, 2019-12-17:
> 
> 
> Quick fix: add a modifier volume for the whole object but the first layers (create a big box and move it a bit in the z axis) and set "bottom perimeters" to 0 for this volume.

In vase mode not possible.
[image]

supermerill, 2019-12-17:
erf, my bad.

Jeroen6, 2019-12-21:
Vase mode seems to assume the bottom is one solid. When you add holes things go wrong. 
In my example the hole is complete ignored:
[Lampshade.zip](https://github.com/prusa3d/PrusaSlicer/files/3991429/Lampshade.zip)

bubnikv, 2020-03-11:
We have improved handling of spiral vase print in PrusaSlicer 2.2.0. Would you please retest with PrusaSlicer-2.2.0-rc2?

From PrusaSlicer 2.2.0-beta change log:
Fixed regression in spiral vase after recent refactoring. Spiral vase now requires reslicing of the model when enabled / disabled as different slicing strategy is newly used for the spiral vase mode to always produce a single contour per layer.

ochm, 2020-03-11:
Looks much better, but center hole in bottom layers is missing in sliced object.. In vase mode and in normal mode too.

bubnikv, 2020-03-12:
Yeah, the spiral vase mode requirements are not very well defined, we are finding them along the way.

In 2.2.0 I have modified the slicing algorithm to close holes and keep just the largest contour always, while it seems the correct solution is to not do that on bottom layers. And the code complexity only grows.

bubnikv, 2020-03-12:
I am not brave enough to hack it just before the final release, so it has to wait for the next release.

kbehren, 2020-05-23:
In 2.2.0 there is no bridging, the hole is just completely ignored and printed solid.
Cura 4.6.1 does a better job slicing vases like these, but has its own issues like not allowing the complete 210 mm in Y direction.

kbehren, 2020-05-26:
@bubnikv did you look into this after the last release?

DarKOrange75, 2020-09-26:
Any news on this issue? I have a model with a big hole in the bottom layer which is completely ignored by Prusa Slicer

kbehren, 2020-09-28:
I don't think there has been any progress. I tried to ping @bubnikv before since he mentioned he could change the behavior but didn't want to do it close to the last release.

floorheight, 2020-12-04:
Confirming this issue still affects vase mode slicing.

**Software:**
Version: PrusaSlicer 2.3.0-beta1
OS + version: Debian 10 (x86_64)
Locale: EN
Config: AppImage - Default runtime parameters, vase mode selected.
Machine: Creality CR10s

**Behaviour:**
Test object: Plant pot (170mm) taken from https://www.thingiverse.com/thing:2394778
Expected: Successful vase mode print with four holes in bottom layer(s), as per model.
Outcome: Solid infill on bottom layer(s), no holes in layer 1. Tested with various bottom layer amounts - same result.

**Attached Images:**
Viewport rotated to show model interior
[image]
Higher layers hidden for better view of issue
[image]

bubnikv, 2020-12-12:
Fixed in PrusaSlicer 2.3.0-beta3.

closed 2020-12-12 by bubnikv

philipbjorge, 2020-12-16:
@bubnikv -- Thanks for fixing this! My girlfriend will love the flower pot I'm printing for her :)

UnrealFakes, 2021-10-31:
I'm on 2.3.55 and this problem still exist

bubnikv, 2021-11-03:
This one slices nicely in PrusaSlicer 2.4.0-beta1.

[image]

This one does not slice as one may expect, but the model is not a valid solid, it is an open surface.

[image]

If fixed with Netfabb, all the holes are closed

[image]

bubnikv, 2021-11-03:
The clay pot model is a valid solid and it is now sliced correctly indeed

[image]

bubnikv, 2021-11-03:
Thus the conclusion is, that the vase mode is fixed indeed for correct models.

@f-stefa @UnrealFakes I wonder why you gave thumbs down. This made me thinking that the issue is still there. Would you  please provide models where PrusaSlicer 2.3.3 or 2.4.0-beta still does not slice a vase model with holes in the bottom correctly?

Jeroen6, 2021-12-28:
I can confirm it now works as expected in PrusaSlicer-2.4.0+win64-202112211614