prusa3d/PrusaSlicer #3327: PrusaSlicer Layers issue / slicer related bug

alientek, 2019-12-17:
### Version
2.2.1

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
MK3S / 3.8.1

### Behavior
PrusaSlicer has a big issue, some specific models show inconsistent layers regardless of flow/temp/layer hight and the result ends up like the top example.
[image]

The bottom one was sliced with S3D, and has no issues at all. Has anyone had a similar issue?

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

[Visor2.zip](https://github.com/prusa3d/PrusaSlicer/files/3971259/Visor2.zip)

beeb, 2019-12-17:
Might be useful to provide the sliced gcode from S3D too

MucaroBoricua, 2019-12-22:
I'm experiencing something similar.

closed 2026-09-29 by prusaslicer-bot[bot]