prusa3d/PrusaSlicer #3331: Is it possible to change the amf file without having to rebuild all support definitions?

Heckw, 2019-12-18:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

supermerill, 2019-12-18:
you should use just a couple of words for the title and put your description in the first post of this issue, you can edit it by clicking edit under the "..." menu, just at the right of the smiley.

For your issue, right-clic on the object and click "reload from disk"

neophyl, 2019-12-18:
Merill that doesn't work if the project has been saved as the reference is to the part stored internally rather than to a location on the hdd.  
The usual work around is to right click on the old part, load in the new part so both are part of the same object and then delete the original part.  That leaves all the modifiers that have been carefully placed but the problem with doing this is that the new part then has to be positioned/scaled/rotated to match the original part.  
Which is a pain in the rear usually.

supermerill, 2019-12-18:
mmm, you have to replace the .model that is stored inside the zip archive that is the .3mf with the updated one.

A "reload from file" button next to reload from disk" may be useful, you're right.

Heckw, 2019-12-19:
I agree, that's exactly what we need. Any chance that this will be realized in the next version?

Heckw, 2019-12-19:
> 
> 
> you should use just a couple of words for the title and put your description in the first post of this issue, you can edit it by clicking edit under the "..." menu, just at the right of the smiley.
> 
> For your issue, right-clic on the object and click "reload from disk"

Noted, thanks!

Heckw, 2019-12-19:
Inside the .3mf file there's a 3dmodel.model. I assume this is the file to replace. What format is this file?
Can I take the new .amf file and rename to 3dmodel.model and replace it in the .3mf file?

Heckw, 2019-12-19:
> Merill that doesn't work if the project has been saved as the reference is to the part stored internally rather than to a location on the hdd.
> The usual work around is to right click on the old [...]

That didn't work out for me. When I delete the original part after loading the new verson the supports disappear with the original part!

neophyl, 2019-12-19:
I dont have any amf files so there may be a difference between amf and stl files.  However the workflow I use for stl files is to place a stl model file on the plater as usual.  Place your modifiers, support enforcers etc as usual.

Now right click on the stl model file and select Add Part >Load.  Load the new updated stl file of your choice.  it should appear in the object tree list under the same branch with the original stl model and all the modifiers.   Then select just the original stl model and delete it.  

That results in the new stl and the modifiers being left on the plater.  You have to manually place it in the same position as the original for the modifiers to be in the correct places which depending on its starting orientation can be a pain but its still easier that re doing lots of carefully placed modifiers.

Heckw, 2019-12-19:
I played around a bit. With stl-files I could reproduce your workaround (emphasis on 'work' :-(( ). With amf or 3mf-files I could not. Here the support bodies are deleted with the original part. I really hope this functionality will be reviewed and something practical will be implemented asap. I'm really astonished that such a basic functionality is missing!

lukasmatena, 2019-12-19:
@enricoturri1966 has recently improved reloading from disk a lot. Please, test that function in PrusaSlicer 2.2.0 (an alpha will be released in a few days) and give us feedback. Thanks.

Heckw, 2019-12-19:
Thank! I look forward to installing PrusaSlicer 2.2.0 alpha and testing 'reloading from disk'.

Heckw, 2019-12-22:
I downloaded 2.2.0 alpha2. It's supposed to offer a solution for this case (replace amf-file).
I couldn't find a new and/or better workflow to do this. Does anyone have more information about an intended workflow?

bubnikv, 2021-06-03:
Is it still a problem with PrusaSlicer 2.3.1?

Heckw, 2021-06-05:
No, it's ok now.

closed 2021-06-05 by Heckw