prusa3d/PrusaSlicer #3334: Feature request: reload stl button

liborw, 2019-12-18:
It would be nice to have a **reload stl** button, the current workflow is, delete all add all instances, find out I had made mistake and forgot to export the stl from cad, so delete all add all once more and so on...

Spacemarine2018, 2019-12-18:
Doesn't "right click -> reload from disk" work?

ochm, 2019-12-18:
> 
> 
> Doesn't "right click -> reload from disk" work?

Button is button. :)

liborw, 2019-12-19:
Didn't know, works for me...

lukasmatena, 2019-12-19:
Is it ok to close the issue then? The function is there and I believe that accessing it from the context menu is adequate.
By the way, the function has been recently improved to retain modifiers and transformations, this improvement will be in PrusaSlicer 2.2.0 (an alpha will be released soon).

DominikPalo, 2019-12-23:
Maybe an automatic popup showing: "The STL file has been modified. Do you want to reload it?" would be also helpful and nice ;) That is how it works in most editors/applications.

liborw, 2020-01-06:
For me the context menu is fine, popup would be nice as sometime you just forget and print old version of the model, doesn't help if you forget to export the stl though.

Closing the issues as for me is solved by the context menu.

closed 2020-01-06 by liborw