prusa3d/PrusaSlicer #3335: Accept blanks in gcode file path at post processing scripts

JoeScotty, 2019-12-18:
### Version 2.1
### Operating system type + version Win10 64 bit 1909
### 3D printer brand / version + firmware version (if known) self constructed corexy, repetier firmware 1.03

### Behavior
Postprocessing script does not work if the attached gcode file path has blanks.

The gcode file path is attached automatically to my defined script or command.
The path is where I save my gcode file.
If this path contains one or more blanks, the postprocessing fails, since the argument ends with the first blank.
draft example
Using curl to send the gcode file to the printer in these cases showed errors like 
c:\windows\system32\curl ... c:\users\myself/3D
while the complete command should be like
c:\windows\system32\curl ... c:\users\myself\3D Objects\project\test.gcode

It would be more convenient if blanks in the automatically attached path would be accepted.

JoeScotty, 2020-01-11:
One more information: PS throws an error if the gcode file path has at least one blank in it.
Translated to english it says something like:

PrusaSlicer has detected an error:
Post-processing script "c:\users\myself\3D objects\scripts\Send2printer.bat" on file
c:\users\myself\3D objects\project\test.gcode failed.
Error code: 1

The error happens before the batch file is called.

bubnikv, 2020-03-15:
on windows, the path should support quoting the windows way
https://blogs.msdn.microsoft.com/twistylittlepassagesallalike/2011/04/23/everyone-quotes-command-line-arguments-the-wrong-way/

JoeScotty, 2020-03-15:
Thanks for your reply!
As far as I understand, PrusaSlicer adds the gcode file path at the end of my post processing call.
If the file path of the gcode file contains no space, everything works
if the filepath of the gcode file contains a space, an error occurrs.
I'm no software developer, but I think the format of the added file path is handled by PrusaSlicer software and kindly ask for fixing.

bubnikv, 2020-03-16:
Could you provide your 3mf and your bat file?

I tested under the debugger and it seems everything works fine. If I enter d:\tool as a G-code post processor into PrusaSlicer, internally the following is sent to the Win32 CreateProcess call:

d:\\tool "D:\temp\path with spaces\name with spaces.gcode"

To me it is quoted correctly. Maybe you incorrectly process the quoted argument inside your script?

JoeScotty, 2020-03-16:
Now I tried all 4 cases: filename w or w/o spaces and path w or w/o spaces.
1) Both w/o spaces works as expected, no error, gcode file is sent to printer.
2) filename with spaces produces one error, Prusaslicer Error Code 6:
[image]

3) path with spaces ("d:\Projects_3D\Path with spaces")produces two errors
3-1) first in cmd windows. Please watch the shortened path at the first space!
[image]

3-2) and second PrusaSlicer Error Code 6 again, like above screenshot
   there is no difference if filename also does contain spaces or does not.

The zipped emf file:
[Path_Name_Test.zip](https://github.com/prusa3d/PrusaSlicer/files/4339385/Path_Name_Test.zip)

The bat File:
[Send2Printer.bat.txt](https://github.com/prusa3d/PrusaSlicer/files/4339388/Send2Printer.bat.txt)

The Call command in print settings, output options:
[image]

PrusaSlicer Version: 2.1.0, win64-201909160915
I also changed the GUI language to english: no difference.

And thanks again for your help!

JoeScotty, 2020-03-22:
Now I found the reason for the problem and also a solution:

Why does it happen?
- the gcode-path, which is appended by prusaslicer, contains no quotes if the path contains no blanks. For example d:\projects\testcube.gcode
- the gcode-path, which is appended by prusaslicer, contains quotes if the path contains blanks, like "d:\3D projects\testcube V2.gcode"
- in my batchfile which uses curl for sending the gcode file to the printer, the gcode-path is passed as a parameter, in the batchfile called with %1. Like...
c:\Windows\System32\curl.exe ..... -F "filename=@%1"
As a result we have kind of nested quotes like
c:\Windows\System32\curl.exe ..... -F "filename=@"d:\3D projects\testcube V2.gcode""
which is not allowed and causes my describes errors.

My solution is:
modified my batchfile to (watch the ~):
c:\Windows\System32\curl.exe ..... -F "filename=@%~1"

Maybe it helps somebody else.

closed 2020-03-22 by JoeScotty

bubnikv, 2020-03-22:
Thanks for heads up. I only know that the quoting on Windows is convoluted, but I did not know these details.