prusa3d/PrusaSlicer #3337: Modifier modifies all following objects when complete individual object is set.

mkbel, 2019-12-18:
### Version
PrusaSlicer-2.1.1+linux64-201912101511

### Operating system type + version
Debian Sid AMD64 with XFCE 4.12.5

### 3D printer brand / version + firmware version (if known)
unrelated

### Behavior
Modifier modifies all following objects when complete individual object is set.
* _Steps needed to reproduce the problem_
  * Add several objects.
  * Modify some object setting of first object. For example extrusion width - External perimeters.
* _Expected Results_
  * Change is applied on to one object.
* _Actual Results_
  * Change is applied to all objects, which was added later then object with some modification.

_Is this a new feature request?_
No, it is bug.

#### Project File (.3MF) where problem occurs
This file is not zipped, just rename to blade-vase-sith-1-5_extrusion_width_bug.3mf .
[blade-vase-sith-1-5_extrusion_width_bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3980932/blade-vase-sith-1-5_extrusion_width_bug.3mf.zip)

lukasmatena, 2019-12-19:
This is certainly a bug. It appears that the problem is not the complete individual objects nor spiral vase mode, I got the same problems when I turned both of these off. Apparently the external extrusion width setting is handled incorrectly. Someone will have to look into it.

mkbel, 2019-12-19:
It looks like object modifier "Extrusion Width Perimeters" is mapped to "Extrusion Width External Perimeters". I have found workaround for my case: When change object modifier "Extrusion Width Perimeters" it is applied to object "Extrusion Width External Perimeters" and I can get results I want. Except for the fact that there is no way to modify first layer perimeters width for single object.

mkbel, 2019-12-19:
This issue is present also in version 2.2.0-alpha0+121-linux-x64.

rtyr, 2021-02-04:
Fixed. Closing.

closed 2021-02-04 by rtyr