prusa3d/PrusaSlicer #3338: PrusaSlicer Scale Function Lacks Individual Axis Scaling Using The Keyboard

alwalmsg, 2019-12-19:
### Version - PrusaSlicer Ver. 2.1.1
### Operating system type + version - Windows 10 Pro
### 3D printer brand / version + firmware version - Mk3S    Ver.  3.8.1

The only way I have found to scale a single Axis in this version of PrusaSlicer is to select the object, use the Scale Tool Icon and then drag the axis to change it's scale factor.  This method makes it very difficult to attain an accurate value.

I tried changing a single axis scale factor value in the Object manipulation window.  Changing the one axis scale factor value causes the other axis factor values to change by the same amount which is not what I want to do. 

Example scale factor values:
Original values              x=100 y=100 z=100
Desired values               x=102 y=100 z=100
What I get is                  x=102 y=102 z=102

There is no way that I have found to accurately enter individual axis scale factors with the keyboard without affecting the other two axis scale factor values.

I have spent hours looking on the Web for an answer with no success.
I may be missing something and if that is the case then please explain how to do this operation.

_Is this a new feature request?_  YES

Thanks
Fred

jmd73, 2019-12-19:
Just click the lock icon to the left of the scale boxes (red square added for clarity).

[image]

jmd73, 2019-12-19:
@alwalmsg This is a duplicate of #3263

So a search for scaling would have shown you the above mentioned issue that is also answered. 

Both issues should be closed.

closed 2020-03-15 by bubnikv