prusa3d/PrusaSlicer #3339: [Feature Request] output filename format with object name and number of copies

sgflux, 2019-12-19:
### Version
Version 2.1.1+win64

**My problem:** '**Export G-Code**' generates name (MyPart.gcode) from 3mf-filename (when saved). Not from stl-file that is in the object-list.

**What I nee:** 'object_name(s)' and 'number of copies' per 'object'

**Like this:**

**{number_of_copies_object(0)}X_{filename_object(0)}_{number_of_copies_object(1)}X_{filename_object(1)}**_{nozzle_diameter[0]}n_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode

**Result:**
5X_MyPart1_4X_MyBetterPart2_0.6n_0.3mm_PLA_MK3S_11h3m.gcode.gcode

italocjs, 2019-12-22:
well, its not a bad idea, but i see some possible problems there, like getting the file name too large for some boards /systems to handle, i had seen this a lot on the past with some older cnc machines and even see it today when i extract stuff with too long names.

sgflux, 2019-12-23:
I use OctoPrint and here I have no problems with long filenames. I jusr tried it with 240 letters: Upload and preview successful.

bubnikv, 2020-02-05:
Also we do not support control loops in our scripting language, so it will be difficult to produce variable number of file names / numbers of instances.

bubnikv, 2020-02-05:
Similar issue https://github.com/prusa3d/PrusaSlicer/issues/2735

closed 2026-09-29 by prusaslicer-bot[bot]